intel/intel-device-plugins-for-kubernetes #1352: Invalid nodeSelector rule for sgxdeviceplugin-sample

MustDie95, 2023-03-09:
**Pre-work:**
1. Deployed Node Feature Discovery using Helm chart version  0.12.1 (https://kubernetes-sigs.github.io/node-feature-discovery/charts)
2. Deployed Intel Device Plugins Operator using Helm chart version 0.26.0 (https://intel.github.io/helm-charts/)
3. Created sgxdeviceplugin-sample using SgxDevicePlugin CRD

**Expected result:** 
- Number of intel-sgx-plugin pods were created and running in inteldeviceplugins-system namespace. 
- All pods with SGX enclaves fully loaded and functional.

**Real world:** 
- No one of intel-sgx-plugin pods created
- SGX enclaves aren't work

**Reason:** 
- Node Selector in sgxdeviceplugin-sample is based on **"intel.feature.node.kubernetes.io/sgx: true"**
label instead of **"feature.node.kubernetes.io/cpu-security.sgx.enabled: true"** created by NFD on working nodes.

**Workaround:**
- Manually create label **"intel.feature.node.kubernetes.io/sgx: true"** on selected nodes.

**Result:** 
- intel-sgx-plugin pods were created and running in inteldeviceplugins-system namespace. 
- All pods with SGX enclaves fully loaded and functional.

mythi, 2023-03-09:
> * Node Selector in sgxdeviceplugin-sample is based on **"intel.feature.node.kubernetes.io/sgx: true"**

It expects users deploy our [`NodeFeatureRules`](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/caa29497e63ee785b23f461c328c2e9868a1d5c0/deployments/nfd/overlays/node-feature-rules/node-feature-rules.yaml#L90-L103) for NFD but I can see it's missing from `INSTALL.md` (if you followed that?)

tkatila, 2023-03-09:
As you are using helm charts for the rest, you could also use sgx helm chart to deploy the CRD. Something like this:
`helm install sgx intel/intel-device-plugins-sgx --set nodeFeatureRule=true`

This also includes the missing NodeFeatureRules.

MustDie95, 2023-03-09:
> As you are using helm charts for the rest, you could also use sgx helm chart to deploy the CRD. Something like this: `helm install sgx intel/intel-device-plugins-sgx --set nodeFeatureRule=true`
> 
> This also includes the missing NodeFeatureRules.

@tkatila Thanks for advice, this was completely resolved my issue! :)

closed 2023-03-09 by MustDie95