intel/intel-device-plugins-for-kubernetes #1353: Update accel-config version for accel-config-demo

vranar-int, 2023-03-10:
accel-config tool has recently been updated to version 4.0
(https://github.com/intel/idxd-config/releases/tag/accel-config-v4.0)

Current docker image for accel-config demo is using 3.5.2.

mythi, 2023-03-10:
> Current docker image for accel-config demo is using 3.5.2.

do you have a particular image in mind where 4.0 is preferred?

mythi, 2023-03-10:
@vranar-int that's good input! 

@ozhuraki PTAL

vranar-int, 2023-03-10:
It is also likely the accel-config tool is used with the device plugin in general so it might be worth checking if an overall update is required. The mentioned image is one place I can 100% confirm because of our use.

eero-t, 2023-04-12:
That uses Debian unstable, but `accel-config` version there is still at 3.5.3.  Debian unstable is rolling distro, so it will update eventually.

mythi, 2023-04-12:
`accel-config-demo` image uses a version we can control and I've asked @ozhuraki to update

closed 2023-05-17 by hj-johannes-lee