intel/intel-device-plugins-for-kubernetes #1354: Submit 0.26 version of Intel Device Plugins Operator to https://operatorhub.io/

mythi, 2023-03-10:


javydekoning, 2023-05-28:
Your helm repo isn't updated to 0.26.1 either. Any reason?

```sh
helm search repo intel
NAME                                	CHART VERSION	APP VERSION	DESCRIPTION
intel/intel-device-plugins-dlb      	0.26.0       	0.26.0     	A Helm chart for Intel DLB Device Plugin
intel/intel-device-plugins-dsa      	0.26.0       	0.26.0     	A Helm chart for Intel DSA Device Plugin
intel/intel-device-plugins-gpu      	0.26.0       	0.26.0     	A Helm chart for Intel GPU Device Plugin
intel/intel-device-plugins-iaa      	0.26.0       	0.26.0     	A Helm chart for Intel IAA Device Plugin
intel/intel-device-plugins-operator 	0.26.0       	0.26.0     	A Helm chart for Intel Device Plugins Operator ...
intel/intel-device-plugins-qat      	0.26.0       	0.26.0     	A Helm chart for Intel QAT Device Plugin
intel/intel-device-plugins-sgx      	0.26.0       	0.26.0     	A Helm chart for Intel SGX Device Plugin
```

tkatila, 2023-05-29:
@javydekoning no particular reason. We are on it.

hj-johannes-lee, 2023-05-31:
https://github.com/k8s-operatorhub/community-operators/pull/2834