intel/intel-device-plugins-for-kubernetes #1355: Error when generate quote for an sgx based application.

xxinran, 2023-03-13:
I am trying to run my own sgx-based application in pod, it fails since it cannot get quote. 

I have pccs service running on my host and I found "Error: No cache data for this platform." in pccs.service log. 

But my application works well on the host, the pccs.service didn't show any error. 

Is it related to sgx plugin? could you please kindly help?

mythi, 2023-03-13:
> I am trying to run my own sgx-based application in pod, it fails since it cannot get quote.

in-proc or out-of-proc quotes. Are you sure your pod ends up using the same PCCS as the app on the host

xxinran, 2023-03-13:
> > I am trying to run my own sgx-based application in pod, it fails since it cannot get quote.
> 
> in-proc or out-of-proc quotes? Are you sure your pod ends up using the same PCCS as the app on the host?

in proc, i have a config map as the following:

```
apiVersion: v1
data:
  sgx_default_qcnl.conf: "{\n  // *** ATTENTION : This file is in JSON format so the
    keys are case sensitive. Don't change them.\n\n  // This sample is a typical config
    file for a development environment which has a local PCCS setup\n  // QPL will
    get PCK certificates as well as quote verification collateral from the local PCCS
    service\n  // The PCCS service uses self-signed certificates\n  // You should
    choose the correct PCCS API version. \"3.1\" will return CRL in raw DER format
    \n  // It is recommended to use \"3.1\" for DCAP 1.12 release and later\n\n  //PCCS
    server address\n  \"pccs_url\": \"https://myip:8081/sgx/certification/v4/\",\n\n
    \ // To accept insecure HTTPS certificate, set this option to false\n  \"use_secure_cert\":
    false,\n\n  \"pccs_api_version\": \"3.1\"\n}\n"
kind: ConfigMap
metadata:
  annotations:
    sgx.intel.com/quote-provider: httpa-pod
  name: inproc-sgx-attestation-conf
```

mythi, 2023-03-13:
> in proc, i have a config map as the following:

and the host has the exact same `/etc/sgx_default_qcnl.conf`?

xxinran, 2023-03-13:
> > in proc, i have a config map as the following:
> 
> and the host has the exact same `/etc/sgx_default_qcnl.conf`?

Not exactly the same, but it has same service address.  I have remove all the commented line in my host's configuration. 
```
{
  "pccs_url": "https://192.168.110.100:8081/sgx/certification/v4/",
  "use_secure_cert": false,
  "pccs_api_version": "3.1",
  "retry_times": 6,
  "retry_delay": 10,
  "pck_cache_expire_hours": 168
}
````

closed 2023-03-15 by xxinran