intel/intel-device-plugins-for-kubernetes #1364: e2e-dlb test seems unstable

tkatila, 2023-03-24:
There seems to be some instability with the e2e-dlb(-gnr) test. For example, my gpu related change failed on dlb twice in a row:
https://github.com/intel/intel-device-plugins-for-kubernetes/actions/runs/4502343804/jobs/7928667288

@hj-johannes-lee can you take a look?

mythi, 2023-03-24:
There's also a crash in the error path that looks suspicious

hj-johannes-lee, 2023-03-24:
> There's also a crash in the error path that looks suspicious

What is 'error path'?

hj-johannes-lee, 2023-03-24:
I found the cause of it:
There was something I could not consider when I made e2e-test to have multiple It()s.

I thought namespace from previous It() is deleted before BeforeEach() that deploys device plugin again.

In other words:
I thought:
namespace is deleted -> device plugin is deployed -> see if resources are available

But, I guess the order for deleting namespace is not guaranteed before BeforeEach().

So, making timeout longer will solve the problem temporarily, but will need to find to make it sure that the plugin (or namespace) is deleted before deploying another for the next run.

mythi, 2023-03-27:
> > There's also a crash in the error path that looks suspicious
> 
> What is 'error path' and what is the error path?

One of the code paths where you end up calling `framework.Failf()`:

```
    [PANICKED]
    Test Panicked
    In [DeferCleanup (Each)] at: /usr/local/go/src/runtime/panic.go:260

    runtime error: invalid memory address or nil pointer dereference

    Full Stack Trace
      k8s.io/kubernetes/test/e2e/framework.(*Framework).dumpNamespaceInfo.func1()
      	/root/workspace/vendor/k8s.io/kubernetes/test/e2e/framework/framework.go:274 +0x5c
      k8s.io/kubernetes/test/e2e/framework.(*Framework).dumpNamespaceInfo(0xc0002ce1e0)
      	/root/workspace/vendor/k8s.io/kubernetes/test/e2e/framework/framework.go:271 +0x139
      reflect.Value.call({0x19a84e0?, 0xc000375b50?, 0x4?}, {0x1d5bfb0, 0x4}, {0x2f427b8, 0x0, 0x56cc57?})
      	/usr/local/go/src/reflect/value.go:586 +0xb07
      reflect.Value.Call({0x19a84e0?, 0xc000375b50?, 0x1?}, {0x2f427b8?, 0x56da15?, 0x0?})
      	/usr/local/go/src/reflect/value.go:370 +0xbc
```

> but will need to find to make it sure that the plugin (or namespace) is deleted before deploying another for the next run.

Let's just fix it properly and avoid workarounds.

PR #1365 (2023-03-30): e2e: device plugin gets deleted before next one is deployed
fixes: #1364
closed 2023-03-30 by mythi