intel/intel-device-plugins-for-kubernetes #1405: Out of date DLB package in dlb demo images

vranar-int, 2023-05-04:
The DLB driver package used in demonstration images is out of date. Current upstream version of the driver is 8.3.0 while the version in the demo images is set to 8.0.0. This causes issues on Sapphire Rapids platform where the demo container terminates with failure with error status stating **DLB2_ST_FEATURE_UNAVAILABLE** even though the dlb.intel.com resource is properly detected by the device plugin in K8s and test pod is properly scheduled.

mythi, 2023-05-04:
We'll see if we can update but since we don't publish this image and you will need to build it too, you can override the existing 8.0.0 version using `--build-arg` for  `DLB_TARBALL` and `DLB_TARBALL_SHA256` to unblock your testing

vranar-int, 2023-05-04:
Thank you for quick response. We'll implement the fix using the `--build-arg` into our test suite.

PR #1412 (2023-05-09): dlb: update demo to use dlb 8.3.0
Fixes: #1405 

e2e test runs successfully with 8.3.0 driver.

With 8.0.0 driver, 'epoll' test fails. So, system should have 8.3.0 driver installed
closed 2023-05-09 by mythi