intel/intel-device-plugins-for-kubernetes #1408: gpu: study and implement a way to limit Pod's i915 count

tkatila, 2023-05-05:
In #1377 it was identified that it would be required in some clusters to limit the Pod's i915 resource count to 1 (or some other value). The idea is to allow setting shared-dev-num to >1 and to prevent users from accessing more GPU resource than designed.

Webhook might be a good way to implement this, but it would be good to study other solutions as well.

eero-t, 2023-05-05:
K8s supports limit ranges: https://kubernetes.io/docs/concepts/policy/limit-range/

@uniemimu thought that it should support extended resources in addition to core ones, but somebody needs to check (also) whether it actually works when one omits default and min values (i.e. whether it allows pods that do not request a GPU):
```
apiVersion: v1
kind: LimitRange
metadata:
  name: gpu-count-constraint
spec:
  limits:
  - max:
      gpu.intel.com/i915: 1
    type: Container
```
@uMartinXu can you try whether that does what you wanted?

vbedida79, 2023-05-08:
Hi @eero-t, tried limitrange. supports extended resources.
It does not accept pods with `gpu.intel.com/i915 > 1`.  
`Warning  FailedCreate  2m55s  job-controller  Error creating: pods "intel-dgpu-clinfo-gphs2" is forbidden: maximum gpu.intel.com/i915 usage per Container is 1, but limit is 3
`
In case of a pods that do not request gpu resource- whether scheduled on GPU node or others, the pod does allocate i915 resource requests and limits:
```
LimitRanger plugin set: gpu.intel.com/i915 request for container httpd;
      gpu.intel.com/i915 limit for container httpd
```
```
containers:
    - resources:
        limits:
          gpu.intel.com/i915: '1'
        requests:
          gpu.intel.com/i915: '1'

```

eero-t, 2023-05-09:
Thanks for testing!

Will it still try to add GPU resource requests for non-GPU pods, if you add:
```
min:
      gpu.intel.com/i915: 0
```
?

vbedida79, 2023-05-09:
If default values are added to limitrange it is added to non GPU pods. If no default values are added to limit range, it allocates below values to non GPU pods. Is this expected?
```
- resources:
        limits:
          gpu.intel.com/i915: '1'
        requests:
          gpu.intel.com/i915: '1'
```

eero-t, 2023-05-09:
> If default values are added to limitrange it is added to non GPU pods.

so if you specify default as `gpu.intel.com/i915: 0`, is the resource request zero for non-GPU pods?

vbedida79, 2023-05-09:
Yes works as expected. If default is` gpu.intel.com/i915: 0` it shows the same for non GPU pods

vbedida79, 2023-05-16:
With max and min in limitrage, for GPU pods requesting resource > 1, it has a forbidden error. 
After adding default limit and request both 0 in limitrange- for non GPU pods, it just shows up as gpu.intel.com/i915: 0 in the spec.

eero-t, 2023-05-17:
> After adding default limit and request both 0- for non GPU pods, it just shows up as `gpu.intel.com/i915: 0` in the spec.

@uniemimu You've been looking more into scheduler.  Do you see any practical problem with zero extended resource request being added to non-GPU processes 8by `limitRange`)?

I.e. does it just look funny, but work fine in practice?

vbedida79, 2023-05-17:
Also, as limitrange is namespace scoped- we only add it in the namespace where workloads are deployed right?

eero-t, 2023-05-22:
Yes, to namespaces where your cluster config RBAC rules allow given k8s users access to GPU resources.

vbedida79, 2023-05-22:
Got it, thanks. 
> I.e. does it just look funny, but work fine in practice?

Apart from this, can we assume limitrange as an efficient solution? Then we could add the yaml in the [project](https://github.com/intel/intel-technology-enabling-for-openshift) 1.0.0 GA as deployment step after GPU deviceplugin is created- for publishing certified operator on OCP 4.12.

eero-t, 2023-05-22:
Does operator component know which namespaces in given cluster are allowed to access GPU resources?   Aren't such RBAC rules rather specific to cluster?

vbedida79, 2023-05-23:
For openshift, there might be some roles to not deploy workloads/objects in `"openshift-"` namespaces created during the cluster. Will check

vbedida79, 2023-05-24:
On openshift these namespaces: `default, kube-system, kube-public, openshift-node, openshift-infra, openshift` do not allow assigning SCC, so its recommended not deploy pods in these namespaces

vbedida79, 2023-05-30:
Any other possible solutions you would suggest to try apart from limitrange?

eero-t, 2023-05-31:
Not that I found of.  And LimitRange seems to be explicitly designed for cases like this.  It just needs to support zero minimum value better (not add request for zero resource).

Or do you think it should have also an option for limit being cluster wide instead of namespace specific?

vbedida79, 2023-06-01:
Thanks, agree with LimitRange. Workloads can be deployed to a specific namespace for individual GPU access. @uMartinXu any thoughts?

uMartinXu, 2023-06-07:
The limitation of the i915 to 1 should be enforced on the whole cluster scope and should not only be applied to the specific namespace.

tkatila, 2023-06-09:
> * Add option to GAS for specifying allowed range for GPU resource requests

GAS would be a possible place for limiting the i915 resource requests, but that would then require using GAS in general.

> * Add option to GPU plugin for rejecting GPU (count) requests outside of given range

I doubt that this is an option as it's quite late in the Pod scheduling flow. I tried returning an error for the `Allocate()` and scheduler just kept retrying leaving multiple `UnexpectedAdmissionError` pods behind. Though the documentation indicates that it's possible somehow more peacefully to return an error.

eero-t, 2023-12-11:
I just noticed that `ResourceQuota` supports also external resources: https://kubernetes.io/docs/concepts/policy/resource-quotas/#resource-quota-for-extended-resources

That could also be experimented with, whether it works any better than `LimitRange` for limiting GPU usage to only specific namespaces.

mythi, 2023-12-11:
Related are the old backlog items #598 and #486

closed 2026-02-13 by tkatila