intel/intel-device-plugins-for-kubernetes #1418: reconciler: incomplete log output

mythi, 2023-05-15:
this line 
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/d60419e4f0063023ea3953c25336c0b2ee9eb9f8/pkg/controllers/reconciler.go#L233

is causing incomplete log lines that look suspicious especially when the diff is empty:
```
I0515 14:13:19.142299 1922029 reconciler.go:233] "" controller="dsadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="DsaDevicePlugin" DsaDevicePlugin="dsadeviceplugin-test" namespace="" name=
"dsadeviceplugin-test" reconcileID=d9fa9595-5651-4796-9c6d-75073106860e ="(MISSING)"
```

hershpa, 2023-05-24:
We have seen similar behavior on OpenShift 4.12

`I0524 15:35:03.613495 1 reconciler.go:233] "intel-device-plugins-manager: " controller="sgxdeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="SgxDevicePlugin" SgxDevicePlugin="sgxdeviceplugin-sample" namespace="" name="sgxdeviceplugin-sample" reconcileID=a30409d8-f114-415d-87e4-7be8b43e54e3 ="(MISSING)"
I0524 15:35:03.613575 1 reconciler.go:233] "intel-device-plugins-manager: " controller="qatdeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="QatDevicePlugin" QatDevicePlugin="qatdeviceplugin-sample" namespace="" name="qatdeviceplugin-sample" reconcileID=72caa4a3-4e4e-40a1-8984-96847a09d204 ="(MISSING)"
I0524 15:35:03.613687 1 reconciler.go:233] "intel-device-plugins-manager: " controller="gpudeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="GpuDevicePlugin" GpuDevicePlugin="gpudeviceplugin-sample" namespace="" name="gpudeviceplugin-sample" reconcileID=51c54915-9f99-47dc-9fe7-bb769a19a4a2 ="(MISSING)"
I0524 15:35:03.627115 1 reconciler.go:233] "intel-device-plugins-manager: " controller="qatdeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="QatDevicePlugin" QatDevicePlugin="qatdeviceplugin-sample" namespace="" name="qatdeviceplugin-sample" reconcileID=a18294f2-8169-4570-9cb8-d13131b319e0 ="(MISSING)"`

tkatila, 2023-10-11:
I revisited this and created a [PR](https://github.com/intel/intel-device-plugins-for-kubernetes/pull/1561) which should decrease those prints.

hershpa, 2023-10-11:
Thanks @tkatila this will definitely help eliminate unwanted confusion.

PR #1561 (2023-10-24): Misc fixes
Fixes #1418 
Fixes #1556 
Fixes #1483 
And bumps the operator min version.
closed 2023-10-24 by mythi