intel/intel-device-plugins-for-kubernetes #1453: failed to build device plugin operator image from dockerfile

Ruoyu-y, 2023-06-16:
I am trying to build the device plugin operator image from the dockerfile under build/docker/intel-deviceplugin-operator.Dockerfile. But i met the error in docker build:

```
 => ERROR [builder 5/5] RUN install -D /intel-device-plugins-for-kubernet  0.2s
------
 > [builder 5/5] RUN install -D /intel-device-plugins-for-kubernetes/LICENSE /install_root/licenses/intel-device-plugins-for-kubernete                s/LICENSE     && if [ ! -d "licenses/operator" ] ; then     GO111MODULE=on go run github.com/google/go-licenses@${GOLICENSES_VERSION}                 save "./cmd/operator"     --save_path /install_root/licenses/operator/go-licenses ;     else mkdir -p /install_root/licenses/operator/                go-licenses/ && cd licenses/operator && cp -r * /install_root/licenses/operator/go-licenses/ ; fi:
#0 0.204 go: github.com/google/go-licenses@: version must not be empty
------
intel-deviceplugin-operator.Dockerfile:52
--------------------
  51 |     RUN (cd cmd/${CMD}; GO111MODULE=${GO111MODULE} CGO_ENABLED=0 go install "${BUILDFLAGS}") && install -D /go/bin/${CMD} /inst                all_root${EP}
  52 | >>> RUN install -D ${DIR}/LICENSE /install_root/licenses/intel-device-plugins-for-kubernetes/LICENSE \
  53 | >>>     && if [ ! -d "licenses/$CMD" ] ; then \
  54 | >>>     GO111MODULE=on go run github.com/google/go-licenses@${GOLICENSES_VERSION} save "./cmd/$CMD" \
  55 | >>>     --save_path /install_root/licenses/$CMD/go-licenses ; \
  56 | >>>     else mkdir -p /install_root/licenses/$CMD/go-licenses/ && cd licenses/$CMD && cp -r * /install_root/licenses/$CMD/go-li                censes/ ; fi
  57 |     ###
--------------------
ERROR: failed to solve: process "/bin/sh -c install -D ${DIR}/LICENSE /install_root/licenses/intel-device-plugins-for-kubernetes/LICEN                SE     && if [ ! -d \"licenses/$CMD\" ] ; then     GO111MODULE=on go run github.com/google/go-licenses@${GOLICENSES_VERSION} save \"./                cmd/$CMD\"     --save_path /install_root/licenses/$CMD/go-licenses ;     else mkdir -p /install_root/licenses/$CMD/go-licenses/ && cd                 licenses/$CMD && cp -r * /install_root/licenses/$CMD/go-licenses/ ; fi" did not complete successfully: exit code: 1
```
I checked the dockerfile and find that there's no value assigned for `GOLICENSES_VERSION`, which caused this error. Any default value shall be set for this argument?

tkatila, 2023-06-16:
We are using makefile to set everything up: `make intel-deviceplugin-operator`

Ruoyu-y, 2023-06-16:
> We are using makefile to set everything up: `make intel-deviceplugin-operator`

Thanks for answering!