intel/intel-device-plugins-for-kubernetes #1457: prepare device plugin API changes for CDI injection

mythi, 2023-06-19:
Kubernetes `KEP4009: Add CDI Devices to device plugin API` got approved. 

It will be added as an alpha feature but we could look into adding the support to our framework and see what use-cases it enables (e.g., FPGA deallocate)

bart0sh, 2024-01-11:
Considering that CDI support for Device Plugins has been graduated to Beta in Kubernetes 1.29, this task becomes more important. Moreover, it can be a show stopper for graduating CDI support for DP to GA as this requires [Gathering feedback from at least 2 device plugin vendors that CDI support works for them](https://github.com/kubernetes/enhancements/tree/master/keps/sig-node/4009-add-cdi-devices-to-device-plugin-api#beta-to-ga-graduation).

@mythi @tkatila Can we prioritise this issue?

tkatila, 2024-01-11:
@bart0sh patches are welcome. :)

Is there a time frame for the feedback?

bart0sh, 2024-02-15:
I'm going to add CDI support for FPGA plugin in near future.

bart0sh, 2024-05-22:
@tkatila 
> Is there a time frame for the feedback?

I'm sorry for missing your question. We're hoping to get [CDI support for device plugins](https://github.com/kubernetes/enhancements/tree/master/keps/sig-node/4009-add-cdi-devices-to-device-plugin-api) graduated to GA in [the next Kubernetes release](https://www.kubernetes.dev/resources/release/).
It would help a lot to achieve this if [this PR](https://github.com/intel/intel-device-plugins-for-kubernetes/pull/1745) is merged.

When it's merged, I'll update the feedback info of [the KEP](https://github.com/kubernetes/enhancements/tree/master/keps/sig-node/4009-add-cdi-devices-to-device-plugin-api) and you can add your feedback there as review comments. But first we should merge the implementation.

mythi, 2024-05-29:
closed by #1745

closed 2024-05-29 by mythi