intel/intel-device-plugins-for-kubernetes #1462: Step1: Call sgx_qe_get_target_info:Error in sgx_qe_get_target_info. 0xe001

wonbin2011, 2023-06-24:
hi , 

when i run   the sgx-sdk-demo   out-of-proc:
`
kubectl pply -k intel-device-plugins-for-kubernetes-0.23.0/deployments/sgx_enclave_apps/overlays/sgx_ecdsa_aesmd_quote/ -n sgx-ecdsa-quote
`
and the result :
[image]

[image]

out of proc:
[image]

Any pointers are greatly appreciated! Thank you!!

wonbin2011, 2023-07-01:
> 

sorry to reply lately.  I thought  inteldeviceplugin version 0.23.0  corresponds  to cluster version 1.23, maybe i am wrong,
is there any key word  that releated to intel sgx ?

wonbin2011, 2023-07-01:
@tkatila 

dmesg log:
[image]
[image]

mythi, 2023-07-18:
@wonbin2011 I believe your setup is missing the PCCS configuration. Quote generation needs to talk to a PCCS.

wonbin2011, 2023-07-19:
> @wonbin2011 I believe your setup is missing the PCCS configuration. Quote generation needs to talk to a PCCS.

could please provide any  infomation about setting up pccs ?

mythi, 2023-07-19:
Intel's DCAP docs are here:
https://download.01.org/intel-sgx/sgx-dcap/1.16/linux/docs/

wonbin2011, 2023-07-19:
> Intel's DCAP docs are here: https://download.01.org/intel-sgx/sgx-dcap/1.16/linux/docs/

thank you  very much

mythi, 2023-08-01:
@wonbin2011 did you get the issue fixed? I'm planning to close this since it's not device plugins related.

closed 2023-08-03 by mythi