intel/intel-device-plugins-for-kubernetes #1471: iaa provisioning issue

mregmi, 2023-07-10:
We are using release 0.27.0 and trying to provision IAA in a SPR machine. We are getting following error in intel-deviceplugin-operator log. The iaa plugin pod is also not getting created. Is there anything we are missing on provisioning steps?

```
E0710 22:33:32.674500 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=050b4279-be1b-42fd-a93c-646acf3cc522
E0710 22:33:32.674533 1 controller.go:324] "intel-device-plugins-manager: Reconciler error" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=050b4279-be1b-42fd-a93c-646acf3cc522
E0710 22:33:32.682214 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=ff45d077-07ea-4732-afed-51b24916875b
E0710 22:33:32.682260 1 controller.go:324] "intel-device-plugins-manager: Reconciler error" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=ff45d077-07ea-4732-afed-51b24916875b
E0710 22:33:32.694577 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=708db220-7e75-4fe0-ab5c-741e628beed9
E0710 22:33:32.694616 1 controller.go:324] "intel-device-plugins-manager: Reconciler error" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=708db220-7e75-4fe0-ab5c-741e628beed9
E0710 22:33:32.717746 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=a4f6ffbc-153d-4318-891c-7793b9f69e6e
E0710 22:33:32.717778 1 controller.go:324] "intel-device-plugins-manager: Reconciler error" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=a4f6ffbc-153d-4318-891c-7793b9f69e6e
E0710 22:33:32.760380 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=6e1bc11c-b4bd-440f-b433-9b54fa60cf9f
E0710 22:33:32.760423 1 controller.go:324] "intel-device-plugins-manager: Reconciler error" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=6e1bc11c-b4bd-440f-b433-9b54fa60cf9f
E0710 22:33:32.843448 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=e30733bf-f359-4487-98e8-dace71490250
E0710 22:33:32.843483 1 controller.go:324] "intel-device-plugins-manager: Reconciler error" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=e30733bf-f359-4487-98e8-dace71490250
E0710 22:33:33.006133 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=863248d7-b6c2-47c9-bdea-ebfadd92a344
E0710 22:33:33.006166 1 controller.go:324] "intel-device-plugins-manager: Reconciler error" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=863248d7-b6c2-47c9-bdea-ebfadd92a344
E0710 22:33:33.328849 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=3f967ff0-93b3-4899-a410-223e9e870821
E0710 22:33:33.328882 1 controller.go:324] "intel-device-plugins-manager: Reconciler error" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=3f967ff0-93b3-4899-a410-223e9e870821
E0710 22:33:33.984672 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil> [... cut ...]
E0710 22:33:53.199282 1 controller.go:324] "intel-device-plugins-manager: Reconciler error" err="daemonsets.apps \"intel-iaa-plugin\" is forbidden: cannot set blockOwnerDeletion if an ownerReference refers to a resource you can't set finalizers on: , <nil>" controller="iaadeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="IaaDevicePlugin" IaaDevicePlugin="iaadeviceplugin-sample" namespace="" name="iaadeviceplugin-sample" reconcileID=6a7a89d8-1033-464f-992b-51b6e47d4572
E0710 22:34:13.695729 1 reconciler.go:456] "intel-device-plugins-manager: unable to create DaemonSet" err="daemonsets.apps \"i
```

mregmi, 2023-07-12:
worked after adding the RBAC role below:

+- apiGroups:
+  - deviceplugin.intel.com
+  resources:
+  - iaadeviceplugins/finalizers
+  verbs:
+  - update

mythi, 2023-07-18:
@mregmi thanks for figuring out the issue, I will submit the fix

hershpa, 2023-07-20:
> worked after adding the RBAC role below in deployments/operator/rbac/role.yaml:
> 
> ```
> +- apiGroups:
> +  - deviceplugin.intel.com
> +  resources:
> +  - iaadeviceplugins/finalizers
> +  verbs:
> +  - update
> ```

@mregmi, how did you figure out that was needed? Might be useful later on :)

PR #1475 (2023-07-20): operator: add missing IaaDevicePlugin finalizers RBAC
Fixes: #1471
closed 2023-07-20 by tkatila