intel/intel-device-plugins-for-kubernetes #1480: qat: add support for device heartbeat status

mythi, 2023-07-24:
https://git.kernel.org/pub/scm/linux/kernel/git/herbert/cryptodev-2.6.git/commit/?id=359b84f8db942ef46d24de8aa397790c3fae22e0

tkatila, 2023-10-06:
What level of support would it be? Read status and report device Healthiness based on it? Allow changing the heartbeat interval via configmap option?

mythi, 2023-10-09:
> Read status and report device Healthiness based on it? 

Yes, each scan checks for -1 in `status` and sets pluginapi `Health` accordingly.

> Allow changing the heartbeat interval via configmap option?

I think not, at list for starters. The interval is fixed on `4xxx` and if configuration is needed, a custom initcontainer could do it.

PR #1559 (2023-10-13): qat: heartbeat support + new capabilities 
Fixes #1480
Fixes #1476
closed 2023-10-13 by mythi