intel/intel-device-plugins-for-kubernetes #1500: enable updateStrategy

mythi, 2023-08-09:
our daemonsets do not have `updateStrategy` defined/enabled.

- [ ] enable default strategy
- [ ] define user configurable bits and add to the operator?

hj-johannes-lee, 2024-07-09:
1. DaemonSet.Spec.UpdateStrategy.Type's default value is RollingUpdate.
Could you explain what you mean that we need to set a default value? Don't all DaemonSets have default value of `updateStrategy` as `RollingUpdate`?

2. > Does your operand use a rolling deployment strategy?
from https://sdk.operatorframework.io/docs/overview/operator-capabilities/

Doesn't 'rolling deployment strategy' mean RollingUpdate of UpdateStrategy type?

3. What does the 'bit' have to change? UpdateStrategy.Type can have 'RollingUpdate' and 'OnDelete'. Do you mean about this? Or something else?

mythi, 2024-07-22:
> 1. Could you explain what you mean that we need to set a default value? Don't all DaemonSets have default value of `updateStrategy` as `RollingUpdate`?

As far as I understand it, the operator SDK expects the operator to *explicitly* set the values. Do you read it differently?

> 3\. What does the 'bit' have to change? `UpdateStrategy.Type` can have `RollingUpdate` or `OnDelete`. Do you mean about this? Or something else?

`RollingUpdate` has parameters that can be configured. They can be useful but perhaps we don't need to add them right now but only focus on making sure we are 'Full lifecycle' compliant. (note: I updated the original description a bit to clarify these)

PR #1786 (2024-07-25): add updateStrategy to daemonSets
closes #1500

moved from #1781
closed 2024-07-25 by hj-johannes-lee