intel/intel-device-plugins-for-kubernetes #1502: runc: /proc/self/attr/keycreate: invalid argument

phhutter, 2023-08-12:
I have tried to deploy the intel sgx plugin (intel-sgx-plugin:0.24.1) on an OpenShift 4.10.41 (Kubernetes; v1.23.12+7566c4d) cluster.
Unfortunately, the intel-sgx-initcontainer fails with the following error message:
```
runc create failed: unable to start container process: error during container init: write /proc/self/attr/keycreate: invalid argument
```

current pod resource definition generated by the `intel-sgx-plugin` DaemonSet
```
kind: Pod
apiVersion: v1
metadata:
  generateName: intel-sgx-plugin-
  annotations:
[..]
    openshift.io/scc: privileged
  name: intel-sgx-plugin-cxnb5
  namespace: intel-sgx-plugin
  ownerReferences:
    - apiVersion: apps/v1
      kind: DaemonSet
      name: intel-sgx-plugin
      controller: true
      blockOwnerDeletion: true
  labels:
    app: intel-sgx-plugin
[..]
spec:
  nodeSelector:
    intel.feature.node.kubernetes.io/sgx: 'true'
  initContainers:
    - name: intel-sgx-initcontainer
      image: '<my-repo>/intel/intel-sgx-initcontainer:0.24.1'
      securityContext:
        seLinuxOptions:
          type: container_device_plugin_init_t
        readOnlyRootFilesystem: true
[..]
  serviceAccountName: default
[..]
  securityContext: {}
  containers:
    - resources: {}
      terminationMessagePath: /dev/termination-log
      name: intel-sgx-plugin
      securityContext:
        seLinuxOptions:
          type: container_device_plugin_t
        readOnlyRootFilesystem: true
        allowPrivilegeEscalation: false
      imagePullPolicy: IfNotPresent
      volumeMounts:
        - name: kubeletsockets
          mountPath: /var/lib/kubelet/device-plugins
        - name: sgxdevices
          readOnly: true
          mountPath: /dev/sgx
        - name: sgx-enclave
          readOnly: true
          mountPath: /dev/sgx_enclave
        - name: sgx-provision
          readOnly: true
          mountPath: /dev/sgx_provision
        - name: kube-api-access-sl66c
          readOnly: true
          mountPath: /var/run/secrets/kubernetes.io/serviceaccount
      terminationMessagePolicy: File
      image: '<my-repo>/intel/intel-sgx-plugin:0.24.1'
      args:
        - '-v'
        - '4'
        - '-enclave-limit'
        - '1'
        - '-provision-limit'
        - '20'
  serviceAccount: default
  volumes:
    - name: kubeletsockets
      hostPath:
        path: /var/lib/kubelet/device-plugins
        type: ''
    - name: sgxdevices
      hostPath:
        path: /dev/sgx
        type: DirectoryOrCreate
    - name: sgx-enclave
      hostPath:
        path: /dev/sgx_enclave
        type: CharDevice
    - name: sgx-provision
      hostPath:
        path: /dev/sgx_provision
        type: CharDevice
    - name: nfd-source-hooks
      hostPath:
        path: /etc/kubernetes/node-feature-discovery/source.d/
        type: DirectoryOrCreate
    - name: kube-api-access-sl66c
[..]
  tolerations:
[..]
status:
  containerStatuses:
    - name: intel-sgx-plugin
      state:
        waiting:
          reason: PodInitializing
      lastState: {}
      ready: false
      restartCount: 0
      image: '<my-repo>/intel/intel-sgx-plugin:0.24.1'
      imageID: ''
      started: false
[..]
  startTime: '2023-08-12T08:25:39Z'
  initContainerStatuses:
    - name: intel-sgx-initcontainer
      state:
        waiting:
          reason: CreateContainerError
          message: >
            container create failed: time="2023-08-12T09:56:13Z" level=error
            msg="runc create failed: unable to start container process: error
            during container init: write /proc/self/attr/keycreate: invalid
            argument"
      lastState: {}
      ready: false
      restartCount: 0
      image: '<my-repo>/intel/intel-sgx-initcontainer:0.24.1'
      imageID: ''
  conditions:
    - type: Initialized
      status: 'False'
      lastProbeTime: null
      lastTransitionTime: '2023-08-12T08:25:39Z'
      reason: ContainersNotInitialized
      message: 'containers with incomplete status: [intel-sgx-initcontainer]'
    - type: Ready
      status: 'False'
      lastProbeTime: null
      lastTransitionTime: '2023-08-12T08:25:39Z'
      reason: ContainersNotReady
      message: 'containers with unready status: [intel-sgx-plugin]'
    - type: ContainersReady
      status: 'False'
      lastProbeTime: null
      lastTransitionTime: '2023-08-12T08:25:39Z'
      reason: ContainersNotReady
      message: 'containers with unready status: [intel-sgx-plugin]'
    - type: PodScheduled
      status: 'True'
      lastProbeTime: null
      lastTransitionTime: '2023-08-12T08:25:39Z'
  phase: Pending
```

I am not sure if the problem has already been solved in a more recent version but I was wondering if it would be possible to upgrade to a newer plugin version when using OpenShift 4.10.x. RedHat officially supports 0.26 (https://catalog.redhat.com/software/container-stacks/detail/61e9f2d7b9cdd99018fc5736), but according to Intel's support matrix, 0.26 should only be used with 
Kubernetes 1.26.

mythi, 2023-08-14:
> I am not sure if the problem has already been solved in a more recent version but I was wondering if it would be possible to upgrade to a newer plugin version when using OpenShift 4.10.x. RedHat [...]

Is your 0.24 deployment using the certified operator stack we initially released for OCP? Searching this error suggests it's SELinux related. 4.10 and 4.11 had some config gaps with SELinux and we temporarily pointed OCP cluster admins to install custom SELinux rules separately and those are now fixed in 4.12+.

phhutter, 2023-08-14:
Hi Mythi,

Thank you very much for your response.

Unfortunately, I couldn't use the official operator from OLM because my current environment requires additional kubernetes resource adjustments. Instead, I used the 0.24.1 release from this upstream repository. I'm happy to contribute my changes back to this repository when I find some time. :-)

Could you please elaborate a bit more on these extra/missing SELinux rules? Were they part of an official workaround provided by RedHat?

Out of curiosity, does Intel already support the SGX intel plugin installation on OCP 4.10? I'm still a bit unsure about your's and RedHat's support matrix.

mythi, 2023-08-15:
> Unfortunately, I couldn't use the official operator from OLM because my current environment requires additional kubernetes resource adjustments. Instead, I used the 0.24.1 release from this upstream repository. I'm happy to contribute my changes back to this repository when I find some time. :-)

Are these enhancements something we could start discussing in a separate issue here? I'd be interested in what changes would be useful.

> Could you please elaborate a bit more on these extra/missing SELinux rules? Were they part of an official workaround provided by RedHat?

We initially released our OCP certified device plugin operator w/ SGX only. It was 4.10 (IIRC) but that release did not have the necessary SElinux rules to run device plugins so we temporarily created https://github.com/intel/user-container-selinux. The necessary fixes were merged in upstream "user-container-selinux" and the most recent OCP releases have the necessary rules in place. @mregmi can comment on the details.

> Out of curiosity, does Intel already support the SGX intel plugin installation on OCP 4.10? I'm still a bit unsure about your's and RedHat's support matrix.

So we initially started releasing our certified device plugin operator when OCP 4.10 was new. Our device plugin operator and the plugins use GA APIs so any new release is expected to work on older k8s versions too. It could be you're hitting the same or related SElinux config gaps we noticed in 4.10 and those are not going to be fixed by moving to a newer device plugin release but needs OCP 4.12 or later.

mythi, 2023-09-04:
closing for now, please re-open if necessary