intel/intel-device-plugins-for-kubernetes #1506: Initcontainer issues with redhat certification

mregmi, 2023-08-22:
RedHat Image certifications for Openshift have added new checks that causes the initcontainers to fail preflight tests.
They have a new rule that modifying core contents from base image is flagged and preflight tests fail.
The initcontainers use toybox and replace all the commands in the image. This is no longer allowed by RedHat in UBI images.
We might have to create a seperate Dockerfile for UBI images ( Openshift ).
Any suggestions on solutions?

mythi, 2023-08-23:
> Any suggestions on solutions?

There are options. If OCP has new enough NFD available (0.13+) we don't need the initcontainer for SGX anymore. I believe you also don't need GPU initcontainer since you have not enabled the fractional resources. GPU is also moving away from the initcontainer completely.

For SGX the caveat is that without the initcontainer, we depend on NFD.

We can of course change the build flow to make toybox installation optional.

mregmi, 2023-08-23:
Thanks. making the toybox optional for UBI sounds like a good solution for those that need initcontainer.

mythi, 2023-08-25:
> Thanks. making the toybox optional for UBI sounds like a good solution for those that need initcontainer.

but note that it won't be available for your 1.0.1 release so you will have to accept that preflight failure now.

mregmi, 2023-08-25:
Yes for this release. we created a custom initcontainer without toybox.

mythi, 2023-09-06:
@mregmi we currently default to `ubi8-micro`. Should that be `ubi9-micro` nowadays?

mregmi, 2023-09-06:
> @mregmi we currently default to `ubi8-micro`. Should that be `ubi9-micro` nowadays?

On RHEL 8.X based systems (ocp 4.12) we are supposed to use ubi8 and ubi9 or ocp 4.13 (RHEL 9.X) and higher. i think we should make it as ubi9 for 0.27 and newer releases.

mythi, 2023-09-06:
> On RHEL 8.X based systems (ocp 4.12) we are supposed to use ubi8 and ubi9 or ocp 4.13 (RHEL 9.X) and higher. i think we should make it as ubi9 for 0.27 and newer releases.

0.28 will default to ubi9

PR #1525 (2023-09-08): images: base image specific toybox config and version updates
Fixes: #1506 

Our *.Dockefile's allow builders to specify FINAL_BASE base image to which static binaries are copied for the final image.

Initcontainers install toybox with a minimum set of "coreutils" needed but with some final base images the minimum set can be even smaller since they may already have /bin/sh, ls etc. installed.

Make initcontainers to have base image specific toybox-config. For now, "static" for distroles/static and "ubi8-micro" for UBI base image configs are created.

At the same time, move the toybox build to  [... cut]
closed 2023-09-08 by tkatila