intel/intel-device-plugins-for-kubernetes #1514: Ubuntu 22.04.3/kernel 6.2.0-31 reporting 0 `gpu.intel.com/i915: "0"`

e3b0c442, 2023-08-31:
Hi all,

I have a cluster with all nodes running Ubuntu 22.04.3, HWE kernel 6.2.0-31.

One node has a discrete Intel Arc A750, and two have 11th-generation Celeron (N5100).

I have installed the Device Plugins Operator, node-feature-discovery, and the GPU device plugin.

All three of the nodes are reporting `gpu.intel.com/i915: "0"`.

My intention is to use these nodes for hardware video transcoding.

Did I miss something? My understanding is specifically for the node with discrete Arc, the 6.2 kernel should have everything needed here. Is there something additional? I tried installing the dkms driver and it failed.

Appreciate any insight/troubleshooting trips here.

tkatila, 2023-08-31:
Hi Nick,

At least the iGPUs on the celeron machines should work out of the box. I'm fairly certain the A750 should work with 6.2.0 also, but it might need force probing. @eero-t you have better insights on the kernel support, can you comment?

As for checking what's wrong with the systems, can you check if the driver is loaded correctly?
`sudo dmesg | grep i915`

If the driver is not found, also check if the driver happens to be compiled in-kernel:
```
$ grep i915 /sys/class/drm/card*/device/uevent
/sys/class/drm/card0/device/uevent:DRIVER=i915
```

Though, if the nodes detect zero i915 resouces, it might be something else. Can you enable verbose printing in the gpu plugin and paste the logs here? The GPU plugin CRD should have `logLevel: 4` by default, but the GPU plugin deployment doesn't. You'd have to edit the plugin Daemonset and add/change the args to say: `-v=4`

eero-t, 2023-08-31:
Force probing is not needed for Arc starting from Linux 6.2: https://www.phoronix.com/review/linux-62-features

Besides dmesg, what's the output of `kubectl describe <nodename> | grep gpu`?

e3b0c442, 2023-08-31:
Hi all,

I really appreciate the quick response here.

A fresh head combined with hints from your questions pointed me to node feature discovery not working properly, causing the nodes not to be labeled such that the plugin started. I double-checked and sure enough, the Intel device plugins rule was missing. After restoring that everything is working as expected.

I'm closing this issue since my problem is resolved. Thanks again!

closed 2023-08-31 by e3b0c442

eero-t, 2023-08-31:
Note that:
* (Even latest) upstream i915 driver is missing some (dGPU) features compared to the out-of-tree backport i915 driver: https://dgpu-docs.intel.com/driver/kernel-driver-types.html
* If you switch to backport i915 DKMS driver (and kernel version supported by it) for Arc, you should take user-space driver packages also from the same repo [1]

[1] Because there are some (dGPU related) kernel uAPI differences between what's used in the backport driver, and what eventually ended up in upstream.  This matters most to media driver which needs to be compiled separately for each uAPI variant, whereas Mesa and compute drivers try to support both uAPIs within same build. For now, you may get better compute perf on dGPUs with the backport driver.  Backport driver validation is more extensive than upstream driver testing, but covers only HW specified on above site, so distro drivers would be still preferred for machines with other HW.