intel/intel-device-plugins-for-kubernetes #1526: SGX device plugin: issues with opening /dev/sgx after maintainence node restart

vtikoo, 2023-09-07:
On upgrading agent pool to AKS 1.25, we are seeing errors with sgx workloads.
The errors are specifically during enclave creation and we have seen two variations so far.

First one is [failure to open the /dev/sgx](https://github.com/intel/linux-sgx/blob/master/psw/urts/linux/edmm_utility.cpp#L113-L116) -
```
/label_exp/ubuntu64/linux-trunk-opensource/psw/urts/linux/edmm_utility.cpp:116] Failed to open Intel SGX device.
```

The second one is an [error during processing](https://github.com/intel/linux-sgx/blob/f656e74afd6ecdbdc2d7397db7ebd46d8cf46ca6/psw/enclave_common/sgx_enclave_common.cpp#L278-L283) of the `sgx_enclave_create` function -

[image]

We were able to mitigate the issues by restarting the associated services. But it would be good to know if there are any gotchas during maintainence node restarts. Specifically if permissions are granted again to SGX device on node restarts.

mythi, 2023-09-07:
> We were able to mitigate the issues by restarting the associated services. But it would be good to know if there are any gotchas during maintainence node restarts. Specifically if SGX device permissions are granted again to workloads on node restarts.

The explanation is fairly simple and it's easy to confirm on your nodes the setup. The "backwards" compatibility of the non-upstream SGX device nodes is slowly going away but some (older) SGX SDK versions have had issues with it.

The SGX device plugin only manages upstream SGX device resources (`/dev/sgx_[enclave|provision]`). We used to support the mode where symlinks from `/dev/sgx/[enclave|provision]` were made available to containers but since 0.25 that was removed.

The implication is that if you have apps only relying on, e.g., `/dev/sgx/enclave` to be present and the device plugin does not make it available to your containers, you will see problems like the first one.

The second one might be Azure specific. The simplest way to address this is to use "out-of-proc" quotes via aesmd rather than the "in-proc" quotes that require the access to `/dev/sgx_provision`. If you're using the plugin from our repo and not the Azure `confcom` add-on, your workloads needs to specify `sgx.intel.com/quote-provider: <name of your container>` annotation to get `/dev/sgx_provision` added to your container. However, the aesmd/"out-of-proc" approach is probably better: `sgx.intel.com/quote-provider: aesmd`

mythi, 2023-09-07:
A couple of pointers: https://github.com/intel/linux-sgx/issues/920, https://github.com/intel/linux-sgx/issues/895, https://github.com/intel/linux-sgx/issues/772

mythi, 2023-09-12:
@vtikoo did I cover your question?

vtikoo, 2023-09-14:
Thanks for the explanation @mythi, this definitely helps. 
We saw the first issue with PSW 2.19, and it looks like that version looks for [both the symlink and the in-kernel driver](https://github.com/intel/linux-sgx/blob/sgx_2.19/psw/urts/linux/edmm_utility.cpp#L102-L116)(in that order). Even if the symlink was not supported, shouldn't the in-kernel driver have been present in the container?

mythi, 2023-09-15:
> Even if the symlink was not supported, shouldn't the in-kernel driver have been present in the container?

Yes, in-kernel nodes are always made available. 

Are you running your workload as non-root user? That might also explain. If the udev-rules are not installed and/or if the default systemd rules for `sgx_enclave` are used, non-root users cannot access these devices.

For the non-root+devices case we have enabled [a contained config ](https://kubernetes.io/blog/2021/11/09/non-root-containers-and-devices/)to have the device nodes inherit per-container uid/gid values.

mythi, 2023-10-09:
@vtikoo I'm closing this since I believe I covered all the questions and there's no bugs we need to fix.

closed 2023-10-09 by mythi