intel/intel-device-plugins-for-kubernetes #1528: documentation: QAT plugin documentation requires some revisions

chaitanya1731, 2023-09-08:
Hello, we recently encountered the documentation for the QAT plugin and noticed that it requires some revisions. Within the [Modes and Configuration Options](https://github.com/intel/intel-device-plugins-for-kubernetes/tree/main/cmd/qat_plugin#modes-and-configuration-options) section, there is still information present about deprecated parameters. Additionally, the resource allocation information includes details about the `/generic` resource but lacks information regarding `/dc` and `/cy`. Do you have any upcoming plans to update the documentation, or is there any revised documentation that we might not be aware of?

mythi, 2023-09-08:
This is the latest we have. The top-level readme has the resources up-to-date. What are the deprecated parameters?

chaitanya1731, 2023-09-08:
Thanks @mythi  can you point to the top-level readme with the resources information? for deprecated parameter, one for example the `-mode` parameter, since this is not in CRD, users cannot set it explicitly, so from the advanced config pov, its confusing for users on how to set it along with other params..

mythi, 2023-09-08:
"Workload authors" section

mythi, 2023-09-11:
> for deprecated parameter, one for example the `-mode` parameter, since this is not in CRD, users cannot set it explicitly, so from the advanced config pov, its confusing for users on how to set it along with other params..

The basic rule for the "sample" CRDs released part of the operator is that it contains all fields so users would already see the available config parameters by looking at the sample. However, I can drop `-mode` from the table.

closed 2023-09-19 by hj-johannes-lee