intel/intel-device-plugins-for-kubernetes #1529: Qat: Ability to do node configurations in cluster with large number of QAT nodes more easily

mregmi, 2023-09-08:
Currently we can configure Node with QAT by creating a configmap with the node name and that gets passed onto the initcontainer and the node is configured.
This does not scale well for a large Clusters with many qat nodes. The users would want to assign say some nodes for compression, some of Crypto and some for mixed.

It would be nice to have a mechanism where the configuration is done with smaller option/configuration and scales with number of nodes. One of the things, we thought about was to use labels. The nodes Labeled for Compression would be configured as compression, Similarly same things could be done for Crypto and mixed. This is just an example, but we things there should be a simpler way to configure that scales with node.

mythi, 2023-09-11:
> It would be nice to have a mechanism where the configuration is done with smaller option/configuration and scales with number of nodes

sounds like a nice-to-have at this point but we could look into allowing cluster admins to create multiple CRs so that `nodeSelector` field can be used to choose which (labeled) nodes the daemonSet is run.

tkatila, 2023-10-13:
Compress setup:
```
kind: ConfigMap
metadata:
  name: qat-config-compress
data:
  qat.conf: ServicesEnabled=dc
---
kind: QatDevicePlugin
metadata:
  name: qat-compress
spec:
  provisioningConfig: qat-config-compress
  nodeSelector:
    my.qat.mode: "compress"
```
Crypto setup:
```
kind: ConfigMap
metadata:
  name: qat-config-crypto
data:
  qat.conf: ServicesEnabled=asym;sym
---
kind: QatDevicePlugin
metadata:
  name: qat-crypto
spec:
  provisioningConfig: qat-config-crypto
  nodeSelector:
    my.qat.mode: "crypto"
```
Cluster admins would label nodes with "my.qat.mode" based on their choice.

Would this work?

mythi, 2023-10-13:
> Would this work?

this was exactly my thinking but we currently allow only one `xDevicePlugin`  resource to be created so it won't work out-of-the box

PR #1579 (2023-11-13): operator: remove one-cr-per-kind limitation
Differentiate objects by adding cr names as suffixes
Drop kind book keeping and related functions from controllers
Make serviceaccounts and clusterrolebindings shared between GPU CRs.
Deny GPU CRs with mixed resource manager selections.

Fixes #1529
closed 2023-11-13 by mythi