intel/intel-device-plugins-for-kubernetes #1532: xpumanager sidecar version issues

vbedida79, 2023-09-12:
Hi, xpumanager sidecar  https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/xpumanager_sidecar/kustomization.yaml uses xpumanager `v1.2.0_golden`. Is there a specific requirement of this version with the sidecar? 
On OCP 4.12, this version fails with `createcontainererror` [xpumanager daemonset](https://github.com/intel/xpumanager/blob/master/deployment/kubernetes/daemonset-intel-xpum.yaml) since the command has changed in the latest intel/xpumanager image.  Daemonset uses the latest image but with v1.2.0 commands. Kustomization.yaml may need a newer version, is that right?
 Is there a specific version xpumanager  recommended to try with sidecar?
Thanks!

vbedida79, 2023-09-13:
Got it, thanks for clarifying. We used xpumanager with v1.2.0_golden with sidecar and GPU plugin 0.26.1 and Flex 140 to get GPU utilization metrics. To avoid, createcontainererror, used 1.2.13 with the dockerhub image with same image tag. But see the issue 64. So for Flex, can I assume just deploying the xpumanager daemonset directly should suffice. Is this correct?

tkatila, 2023-09-13:
> So for Flex, can I assume just deploying the xpumanager daemonset directly should suffice. Is this correct?

Yes, xpumanager-sidecar doesn't add anything for the Flex series.

tkatila, 2023-09-20:
@vbedida79 ok to close this?

vbedida79, 2023-09-20:
yes. thank you

closed 2023-09-20 by vbedida79