intel/intel-device-plugins-for-kubernetes #1542: qat plugin init container crashes with provisioning config

vbedida79, 2023-09-21:
On OCP 4.12.26, when using the provisioning configmap with complete crypto or compress resources the qat plugin's init container crashes. For the first time when using this configuration, there are no issues with plugin. 
If we delete and recreate with same or new resource (cy/dc), we notice this issue. The error is inconsistent and work around is unloading/reloading drivers. The resources not being cleaned up properly- could that be the root cause?

```
16
tee: '/sys/bus/pci/devices/0000:6b:00.0/sriov_numvfs': Invalid argument
16
tee: '/sys/bus/pci/devices/0000:e8:00.0/sriov_numvfs': Invalid argument
```

mythi, 2023-09-22:
@hj-johannes-lee can you reproduce?

hj-johannes-lee, 2023-10-12:
I cannot reproduce it..
Could you just show what you did?

mythi, 2024-02-08:
@vbedida79 we cannot reproduce. is this still an issue on your side?

vbedida79, 2024-02-08:
Checked it on OCP 4.14 OCP, did not notice any error. Thanks. 
Closing this issue.

closed 2024-02-08 by vbedida79