intel/intel-device-plugins-for-kubernetes #1550: Issue with release 0.28.0 with openshift

mregmi, 2023-09-28:
We are building intel device plugin operator for Openshift 4.13.10 but we are having issues deploying the CR. When we try to deploy CR, it gives an error. The errors are same for all plugins (with corresponding endpoint difference).

Any idea whats causing it.

Danger alert:Error
Fix the following errors:
Error "failed calling webhook "msgxdeviceplugin.kb.io": failed to call webhook: Post "https://inteldeviceplugins-controller-manager-service.openshift-operators.svc:443/mutate-deviceplugin-intel-com-v1-sgxdeviceplugin?timeout=10s": dial tcp 10.131.1.249:9443: connect: connection refused" for field "undefined".

mythi, 2023-09-29:
> deploy CR, it gives an error. The errors are same for all plugins (with corresponding endpoint difference).

The only bigger change to webhooks was e92b752d75ea8a7. Can you copy the `MutatingWebhookConfiguration`

`kubectl get mutatingwebhookconfiguration inteldeviceplugins-mutating-webhook-configuration -o yaml`

tkatila, 2023-09-29:
Is this about timing? I tried to apply the operator and the GPU CR back to back:
```
$ kubectl apply -k deployments/operator/default/ && kubectl apply -f deployments/operator/samples/deviceplugin_v1_gpudeviceplugin.yaml
namespace/inteldeviceplugins-system created
..SNIP..
validatingwebhookconfiguration.admissionregistration.k8s.io/inteldeviceplugins-validating-webhook-configuration created
Error from server (InternalError): error when creating "deployments/operator/samples/deviceplugin_v1_gpudeviceplugin.yaml": Internal error occurred: failed calling webhook "mgpudeviceplugin.kb.io": failed to call webhook: Post "https://inteldeviceplugins-webhook-service.inteldeviceplugins-system.svc:443/mutate-deviceplugin-intel-com-v1-gpudeviceplugin?timeout=10s": dial tcp 10.101.21.24:443: connect: connection refused
```

But the apply succeeded when I tried it some seconds after it:
```
$ kubectl apply -f deployments/operator/samples/deviceplugin_v1_gpudeviceplugin.yaml 
gpudeviceplugin.deviceplugin.intel.com/gpudeviceplugin-sample created
```

tkatila, 2023-09-29:
Using a kubectl wait seems to fix it:
```
$ kubectl apply -k deployments/operator/default/ && kubectl wait pods -n inteldeviceplugins-system -l control-plane=controller-manager --for condition=Ready && kubectl apply -f deployments/operator/samples/deviceplugin_v1_gpudeviceplugin.yaml
namespace/inteldeviceplugins-system created
..SNIP..
validatingwebhookconfiguration.admissionregistration.k8s.io/inteldeviceplugins-validating-webhook-configuration created
pod/inteldeviceplugins-controller-manager-7cd949c7d-zptn4 condition met
gpudeviceplugin.deviceplugin.intel.com/gpudeviceplugin-sample created
```

mregmi, 2023-09-29:
Looks like its intermittent. After trying several times, it worked for us. The customers just use the Operator page's UI for this part, so not sure how we can add delay.
Its strange why it suddenly started happening.

mythi, 2023-10-02:
@mregmi did you see a pattern on your side when it's not working and when it is? The command @tkatila just suggests that both containers in the pod must be up (i.e., `2/2`) before setting up any CRs.

mregmi, 2023-10-02:
I will recheck to see if both containers were up. ty

mythi, 2023-10-05:
@mregmi any findings? I believe we could close this?

mregmi, 2023-10-05:
Yes we can close it. I could not reproduce it now. It seems like i tried before all containers were up.

closed 2023-10-09 by mythi