intel/intel-device-plugins-for-kubernetes #1564: GPU: Error response from daemon: invalid volume specification

mzernovx, 2023-10-12:
**Environment:**
- kubernetes 1.27.3
- docker v20.10.20

**Steps to reproduce:**
- Setup Intel Device Plugins
- Create any pod with gpu.intel.com/i915 resource allocated

**Expected behaviour:** pod running

**Actual behaviour:**
pod in CreateContainerError state
`Warning  Failed          2m49s (x12 over 5m3s)  kubelet            Error: Error response from daemon: invalid volume specification: '/dev/dri/by-path/pci-0000:b7:00.0-card:/dev/dri/by-path/pci-0000:b7:00.0-card:ro'`

tkatila, 2023-10-12:
Thanks for reporting this. Did you verify that it's only on docker runtime?

tkatila, 2023-10-12:
The change that is causing this was introduced on 0.26.1 version. You can workaround it by using 0.26.0 in the mean while.

mythi, 2023-10-12:
I remember we have had similar cases with volume mounts where the paths have had colons and docker is used. Is docker mandatory here or could proper CRI runtime be used?

mzernovx, 2023-10-12:
@tkatila I can confirm that with containerd it's working fine.

mzernovx, 2023-10-12:
> I remember we have had similar cases with volume mounts where the paths have had colons and docker is used. Is docker mandatory here or could proper CRI runtime be used?

[BMRA/VMRA](https://github.com/intel/container-experience-kits) uses docker as a default container runtime.

tkatila, 2023-10-13:
> Have you tried any newer Docker version?

I tried a newer version and it reproduces with it:
```
$ dpkg --list | grep Docker
ii  docker-buildx-plugin                             0.11.2-1~ubuntu.22.04~jammy                 amd64        Docker Buildx cli plugin.
ii  docker-ce                                        5:24.0.6-1~ubuntu.22.04~jammy               amd64        Docker: the open-source application container engine
ii  docker-ce-cli                                    5:24.0.6-1~ubuntu.22.04~jammy               amd64        Docker CLI: the open-source application container engine
ii  docker-ce-rootless-extras                        5:24.0.6-1~ubuntu.22.04~jammy               amd64        Rootless support for Docker.
ii  docker-compose-plugin                            2.21.0-1~ubuntu.22.04~jammy                 amd64        Docker Compose (V2) plugin for the Docker CLI.
```
Pod fails with:
```
  Warning  Failed     8s (x2 over 9s)  kubelet            Error: Error response from daemon: invalid volume specification: '/dev/dri/by-path/pci-0000:00:02.0-card:/dev/dri/by-path/pci-0000:00:02.0-card:ro'
```

Docker Engine is mentioned in container runtimes in k8s docs: https://kubernetes.io/docs/setup/production-environment/container-runtimes/#docker that would suggest it's still "ok" to use it.

But to me this is a bug with the docker engine as it works fine with containerd and cri-o. My thought process for this is:
1) File a bug for the docker engine about it not being able to mount paths with `:`.
2) https://github.com/intel/container-experience-kits for docker installation, stick with 0.26.0 GPU plugin
3) If/when the docker engine bug is resolved, update the GPU plugin to the latest version

I do not want to remove the "by-path" mounting as it's required by distributed training. And adding some cli arg or env variable to temporarily disable it feels icky.

tkatila, 2023-10-13:
It seems that a colon in volumes/binds is a known issue:
https://github.com/docker/docker-py/issues/2041
https://github.com/moby/moby/issues/39293
https://github.com/moby/moby/issues/22825

mzernovx, 2023-10-13:
Looks like there's a workaround to use --mount arg with Docker but there's no clear way to utilize this from the side of Kubernetes.

The most suitable fix for this bug seems to be avoiding using /dev/dri/by-path/xxx as they are basically symlinks to devices in /dev/dri

mythi, 2023-10-13:
> The most suitable fix for this bug seems to be avoiding using /dev/dri/by-path/xxx

Avoid using docker is not an option?

mzernovx, 2023-10-13:
> Avoid using docker is not an option?

@mythi BMRA/VMRA still uses docker as a "primary" container runtime. The product is build around customers and their needs, so avoiding using Docker is not an option for us.

stefb69, 2024-10-20:
### **Workaround: Prevent Creation of `/dev/dri/by-path` Symlinks with Colons**

**Issue:**
Docker has a bug that prevents containers from starting when bind-mounting paths containing colons (`:`), such as `/dev/dri/by-path/pci-0000:00:02.0-card`. This affects applications like Plex that require access to `/dev/dri` devices for hardware acceleration.

**Solution:**
Modify the udev rules to prevent the creation of `/dev/dri/by-path` symlinks with colons by replacing them with hyphens (`-`). This ensures compatibility with Docker and Kubernetes by avoiding paths with colons.

**Steps to Implement the Workaround:**

1. **Identify the Culprit udev Rule:**
   
   The symlinks are created by the `60-drm.rules` file, which contains rules for DRM (Direct Rendering Manager) devices.

2. **Create a Custom udev Rule:**
   
   To override the existing rules without modifying system files, create a new udev rule with higher priority.

   ```bash
   sudo nano /etc/udev/rules.d/59-drm-custom.rules
   ```

3. **Add the Following Content to the Custom Rule:**

   ```udev
   # Prevent creation of original by-path symlinks and create modified symlinks without colons

   ACTION!="remove", SUBSYSTEM=="drm", SUBSYSTEMS=="pci|usb|platform", IMPORT{builtin}="path_id"

   # Replace colons with hyphens in ID_PATH
   ENV{ID_PATH}=="?*", PROGRAM="/bin/sh -c 'echo $env{ID_PATH} | sed \"s/:/-/g\"'", ENV{ID_PATH}="%c"

   # Create new symlinks with modified ID_PATH
   ENV{ID_PATH}=="?*", KERNEL=="card*", SYMLINK+="dri/by-path/$env{ID_PATH}-card"
   ENV{ID_PATH}=="?*", KERNEL=="controlD*", SYMLINK+="dri/by-path/$env{ID_PATH}-control"
   ENV{ID_PATH}=="?*", KERNEL=="renderD*", SYMLINK+="dri/by-path/$env{ID_PATH}-render"

   # Stop further processing to prevent original symlinks from being created
   OPTIONS+="last_rule"
   ```

   **Explanation:**
   - **Import `path_id`:** Ensures `ID_PATH` is available.
   - **Replace Colons:** Uses `sed` to replace all `:` with `-` in `ID_PATH`.
   - **Create Modified Symlinks:** Generates new symlinks without colons.
   - **Stop Further Processing:** `OPTIONS+="last_rule"` prevents the original `60-drm.rules` from adding the problematic symlinks.

4. **Set Proper Permissions for the Rule File:**

   ```bash
   sudo chmod 644 /etc/udev/rules.d/59-drm-custom.rules
   ```

5. **Reload udev Rules and Apply Changes:**

   ```bash
   sudo udevadm control --reload-rules
   sudo udevadm trigger --subsystem-match=drm
   ```

6. **Remove Existing Problematic Symlinks (If Any):**

   ```bash
   sudo rm -rf /dev/dri/by-path/*
   sudo udevadm trigger --subsystem-match=drm
   ```

7. **Verify the New Symlinks:**

   Check that the symlinks in `/dev/dri/by-path` no longer contain colons.

   ```bash
   ls /dev/dri/by-path
   ```

   **Expected Output:**
   ```
   pci-0000-00-02.0-card
   pci-0000-00-02.0-render
   ```

**Notes:**
- **Backup Existing Rules:** It's advisable to back up existing udev rules before making changes.
  
  ```bash
  sudo cp /lib/udev/rules.d/60-drm.rules /lib/udev/rules.d/60-drm.rules.bak
  ```

- **System Updates:** Custom rules in `/etc/udev/rules.d/` take precedence over system rules and are not overwritten during system updates.

- **Testing:** After applying the workaround, ensure that Plex and other applications function correctly with the new symlink paths.

**Conclusion:**
By creating a custom udev rule that modifies the `ID_PATH` to exclude colons, Docker and Kubernetes can successfully bind-mount the necessary `/dev/dri` paths without encountering errors. This allows Plex to utilize hardware acceleration as intended.