intel/intel-device-plugins-for-kubernetes #1571: QAT plugin crashes when device is detected, but not fully configured

brgavino, 2023-10-20:
A QAT device may be present on a node, but may only be configured with the kernel driver - or SR-IOV mode may not have been enabled. The operator will do a scan of the available PCI devices but does not check if any of the further pre-requisites as described in the QAT plugin pre-requisites section (possibly, vfio-pci driver loaded, vfs enabled on QAT device, etc). 

This could be the case when a QAT device is present in the system, but will not be available for node resource allocation and exposed to the cluster - but other nodes may have configured QAT devices available.

It is unclear how to limit the deployment of plugins via the operator to avoid  nodes with available, unconfigured devices on the cluster without disabling deployment of QAT plugin to the whole cluster. In this case the operator should continue deployment of other detected devices and avoid attempting to deploy plugins to unconfigured, but installed nodes.

Please advise on correct approach/behavior

Example failure of QAT plugin when device is present via lspci, but not intended to be configured on the node. 
```
I1020 03:28:26.413601       1 qat_plugin.go:62] QAT device plugin started in 'dpdk' mode
E1020 03:28:26.413805       1 manager.go:102] Device scan failed: open /sys/bus/pci/drivers/vfio-pci/new_id: permission denied
write to driver failed: 8086 4941
github.com/intel/intel-device-plugins-for-kubernetes/cmd/qat_plugin/dpdkdrv.writeToDriver
	/go/src/github.com/intel/intel-device-plugins-for-kubernetes/cmd/qat_plugin/dpdkdrv/dpdkdrv.go:445
github.com/intel/intel-device-plugins-for-kubernetes/cmd/qat_plugin/dpdkdrv.(*DevicePlugin).setupDeviceIDs
	/go/src/github.com/intel/intel-device-plugins-for-kubernetes/cmd/qat_plugin/dpdkdrv/dpdkdrv.go:196
github.com/intel/intel-device-plugins-for-kubernetes/cmd/qat_plugin/dpdkdrv.(*DevicePlugin).Scan
	/go/src/github.com/intel/intel-device-plugins-for-kubernetes/cmd/qat_plugin/dpdkdrv/dpdkdrv.go:210
github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin.(*Manager).Run.func1
	/go/src/github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin/manager.go:100
runtime.goexit
	/usr/local/go/src/runtime/asm_amd64.s:1598
failed to set device ID 4941 for vfio-pci. Driver module not loaded?
```

mythi, 2023-10-20:
> It is unclear how to limit the deployment of plugins via the operator to avoid nodes with available, unconfigured devices on the cluster without disabling deployment of QAT plugin to the whole [...]

The operator is not HW aware but it just deploys the device plugin daemonSets based on [user provided resources](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/operator/samples/deviceplugin_v1_qatdeviceplugin.yaml). The CRD two fields relevant to your questions:

`initImage` is used for HW provisioning. It enables SR-IOV if it's not enabled and does other configurations if requested.
`nodeSelector` is the filter label that is used to select the right nodes for the QAT plugin to run. The default label comes node-feature-discovery based on a [`NodeFeatureRule`](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/0ffe7d01ca0bda5d97dd2cb6db1a179d58e6d9f9/deployments/nfd/overlays/node-feature-rules/node-feature-rules.yaml#L77-L88).

eero-t, 2023-10-20:
>> The operator will do a scan of the available PCI devices but does not check if any of the further pre-requisites as described in the QAT plugin pre-requisites section (possibly, vfio-pci driver loaded, vfs enabled on QAT device, etc).
>
> `nodeSelector` is the filter label that is used to select the right nodes for the QAT plugin to run. The default label comes node-feature-discovery based on a [`NodeFeatureRule`](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/0ffe7d01ca0bda5d97dd2cb6db1a179d58e6d9f9/deployments/nfd/overlays/node-feature-rules/node-feature-rules.yaml#L77-L88).

There could be another NFD label rule for VF module (assuming the related features are all configured as modules on distro kernels or loaded as DKMS), and that label can be added to the `nodeSelector` when `dpdkDriver: vfio-pci` option is used.

NFD label rule can also check PCI device `sriov_totalvfs` attribute for the rules:
https://kubernetes-sigs.github.io/node-feature-discovery/v0.14/usage/customization-guide.html#available-features

(On quick check did not find when that was added, but it's there e.g. inNFD v0.12 docs.)

brgavino, 2023-10-20:
Thanks, this is great information to have. Combining this it is possible to ignore certain nodes, and also to ignore nodes that are not configured (such as for sriov). Should this be added to some section of documentation (such as the end for [This section of the README in QAT](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/qat_plugin/README.md#checking-for-hardware))?

eero-t, 2023-10-20:
Can you do some testing with such rules?

If that works fine, project NFD rules overlay can be updated, and QAT example changed to use the new label.

I think the relation between label used in nodeSelector, the driver variant selected in the same YAML file, and what options there are for those, is better documented directly in the example file comments.  README can then have more generic note about that, e.g. "Example file documents relations between labels given for nodeSelector, and the requested QAT attributes".

tkatila, 2023-10-23:
Could we just print the error and not abort? It would increase the feedback loop for cases where a node should get QAT resources but is incorrectly configured.

tkatila, 2023-10-23:
For the NFD rule, we could also add checks for vfio-pci to be either loaded as a module (kernel.loadedmodule) or built-in (kernel.enabledmodule).

brgavino, 2023-10-25:
@eero-t 
> NFD label rule can also check PCI device `sriov_totalvfs` attribute for the rules: https://kubernetes-sigs.github.io/node-feature-discovery/v0.14/usage/customization-guide.html#available-features

Is this going to be a helpful selector? `vfio-pci` may be available and `sriov_totalvfs` set for each pci device, but for whatever reason sriov may not be in enabled state (particularly when in kernel builtin) - `/sys/module/vfio_pci/parameters/enable_sriov` = *N*  (I haven't found a way to check this by features, yet)

There's several states in how sriov/vfio-pci may not be in a configured state needed by the plugin. We can at least hope that with all of these present, QAT plugin can attempt to be successful in initializing, and not render the node unschedulable.
(this doesn't include any rules for nodes to explicitly exclude, just features related to QAT / vfio config state).
[qat-exclusion.yaml.txt](https://github.com/intel/intel-device-plugins-for-kubernetes/files/13168610/qat-exclusion.yaml.txt)

eero-t, 2023-10-25:
I thought `sriov_totalvfs` is max number of VFs a given kernel device supports.  If SR-IOV is not enabled for the device, it being non-zero sounds like a bug...?

mythi, 2023-10-26:
> Is this going to be a helpful selector? `vfio-pci` may be available and `sriov_totalvfs` set for each pci device, but for whatever reason sriov may not be in enabled state (particularly when in [...]

`enable_sriov` for `vfio-pci` is not relevant here. As suggested by @tkatila, we can improve the labeling by adding a rule for vfio-pci being available (module loaded or built-in available). We can also add "IOMMU enabled" which is another pre-condition.

> There's several states in how sriov/vfio-pci may not be in a configured state needed by the plugin. We can at least hope that with all of these present, QAT plugin can attempt to be successful in [...]

I'm not able to follow this part. A plugin failing won't make the node unschedulable (other than workloads requesting the plugin resources won't land there).

mythi, 2023-10-31:
> To summarize: handle explicit "qat plugin exclusion" either with NFD, or via the qat plugin deployment ?

Perhaps use [NFD Tainting Feature](https://kubernetes-sigs.github.io/node-feature-discovery/v0.14/usage/customization-guide.html?highlight=taint#nodefeaturerule-tainting-feature)?.

mythi, 2024-02-14:
@brgavino are you OK we close this or is there something missing still? We have updated the default NFD rules to make the labeling more reliable (e2a7e5d77824ecb759beebd8d892c40b883e3a0e) and the default QAT deployment sets an annotation to make the Apparmor case working (ab0e8bc146ebcd0292d04383b7c019c31db864e0). Also, the NFD Tainting feature was proposed.

brgavino, 2024-02-14:
looks good, thanks for the update - can close this

mythi, 2024-02-14:
@brgavino thanks for the confirmation, closing