intel/intel-device-plugins-for-kubernetes #1575: Improve QAT+Apparmor case

mythi, 2023-10-24:
Since https://github.com/intel/intel-device-plugins-for-kubernetes/issues/381, we've known that adding

```
 annotations:
   container.apparmor.security.beta.kubernetes.io/intel-qat-plugin: unconfined
```

for the QAT plugin daemonSet can be used to mitigate an issue where the plugin fails to initialize on an Apparmor enabled OS.

Triggered by #1571, I'm noticing the annotation setup is poorly documented so users are expected to run into the same problem

We have several ways to improve the case:

- Add the annotation by default (it would be ignored on systems that don't have Apparmor but then we'd loose configurability)
- Move `dpdkDrv` setup to initcontainer OR document how vfio-pci can be automatically made to probe QAT VFs (via `ids` module param)
- Make the issue more visible in the docs.
- ...

ozhuraki, 2023-11-03:
Let's make this annotation default. If the confugurability is needed, the annotation can be dropped with the kustomization.

mythi, 2023-11-06:
> Let's make this annotation default. If the confugurability is needed, the annotation can be dropped with the kustomization.

I thought about making it default too but the problem is that with the operator it cannot be dropped because kustomization is not involved.

PR #1591 (2024-01-10): qat: Add AppArmor unconfided anntotation configurability in the operator
Fixes: #1575
closed 2024-01-10 by mythi