intel/intel-device-plugins-for-kubernetes #1590: GPU: Selinux issue with workload

mregmi, 2023-11-08:
We are seeing a strange SELinux issue with GPU workload. The workload is CLinfo and its is running an non-root. 
When it is trying to access the GPU, the the SELinux denies access to `/dev/dri/renderD128`.  More details are in this issue: https://github.com/intel/intel-technology-enabling-for-openshift/issues/107

After investigating a bit, we root caused it to the `by-path` feature added to gpu plugin. We have verified by reverting this change on an image and it works for us.
I am not sure what exactly is causing this but i suspect, the device file is not being made available for the workload correctly.

mythi, 2023-11-08:
> After investigating a bit, we root caused it to the `by-path` feature added to gpu plugin. We have verified by reverting this change on an image and it works for us.

Is this the same issue we chatted with Dan earlier? The problem was the `by-path` symlink labeling.

mregmi, 2023-11-08:
@mythi Yes. its the same issue.

mythi, 2023-11-08:
My takeaway was a fix was provided. On our side, we have the need to keep `by-path` so the fix likely needs more SELinux changes?

mregmi, 2023-11-08:
Yea its little strange why that lead to SELinux issue. it seems in that change all nodes are added properly. May need some more debugging to know why exactly its causing that.

mregmi, 2023-11-08:
One thing that stands out to me is the files and folders in by-path are labelled differently and also the owner/groups are different. may be that's the issue? Will check what exactly sets the owner and label
`sh-5.1# ls -lZ /dev/dri
total 0
drwxr-xr-x. 2 root root   system_u:object_r:device_t:s0          140 Oct 31 04:50 by-path
crw-rw----. 1 root video  system_u:object_r:dri_device_t:s0 226,   0 Oct 31 03:51 card0
crw-rw----. 1 root video  system_u:object_r:dri_device_t:s0 226,   1 Oct 31 04:50 card1
crw-rw----. 1 root video  system_u:object_r:dri_device_t:s0 226,   2 Oct 31 04:50 card2
crw-rw-rw-. 1 root render system_u:object_r:dri_device_t:s0 226, 128 Oct 31 04:50 renderD128
crw-rw-rw-. 1 root render system_u:object_r:dri_device_t:s0 226, 129 Oct 31 04:50 renderD129
`

mregmi, 2023-11-08:
I think the issue is that labels inside the container is incorrect and is labelled as `dri_device_t` instead of `container_file_t`
Not sure why the label under by-path is wrong.

Workload Pod

```
sh-4.4$ ls -laZ /dev/dri   
total 0
drwxr-xr-x. 3 root root  system_u:object_r:container_file_t:s0:c27,c28      140 Nov  8 22:42 .
drwxr-xr-x. 6 root root  system_u:object_r:container_file_t:s0:c27,c28      380 Nov  8 22:42 ..
drwxr-xr-x. 2 root root  system_u:object_r:container_file_t:s0:c27,c28      120 Nov  8 22:42 by-path
crw-rw-rw-. 1 root video system_u:object_r:container_file_t:s0:c27,c28 226,   1 Nov  8 22:42 card1
crw-rw-rw-. 1 root video system_u:object_r:container_file_t:s0:c27,c28 226,   2 Nov  8 22:42 card2
crw-rw-rw-. 1 root   797 system_u:object_r:container_file_t:s0:c27,c28 226, 128 Nov  8 22:42 renderD128
crw-rw-rw-. 1 root   797 system_u:object_r:container_file_t:s0:c27,c28 226, 129 Nov  8 22:42 renderD129
sh-4.4$ ls -laZ /dev/dri/by-path/
total 0
drwxr-xr-x. 2 root root  system_u:object_r:container_file_t:s0:c27,c28      120 Nov  8 22:42 .
drwxr-xr-x. 3 root root  system_u:object_r:container_file_t:s0:c27,c28      140 Nov  8 22:42 ..
crw-rw----. 1 root video system_u:object_r:dri_device_t:s0             226,   1 Oct 31 04:50 pci-0000:37:00.0-card
crw-rw-rw-. 1 root   797 system_u:object_r:dri_device_t:s0             226, 128 Oct 31 04:50 pci-0000:37:00.0-render
crw-rw----. 1 root video system_u:object_r:dri_device_t:s0             226,   2 Oct 31 04:50 pci-0000:3c:00.0-card
crw-rw-rw-. 1 root   797 system_u:object_r:dri_device_t:s0             226, 129 Oct 31 04:50 pci-0000:3c:00.0-render
sh-4.4$
```

eero-t, 2023-11-09:
I checked host `by-path` dir contents with upstream 6.6 kernel under Ubuntu, 6.5 under Fedora, and 5.14+DKMS under SLES.

On all of those hosts,  `by-path` content indeed differs slightly from Pod contents. All files are:
* symlinks to parent dir files
* Owned by `root:root`

On Fedora it looks like this:
```
$ ls -laZ /dev/dri/by-path/
total 0
drwxr-xr-x. 2 root root system_u:object_r:device_t:s0  80  9.11. 10:44 .
drwxr-xr-x. 3 root root system_u:object_r:device_t:s0 100  9.11. 10:44 ..
lrwxrwxrwx. 1 root root system_u:object_r:device_t:s0   8  9.11. 10:44 pci-0000:00:02.0-card -> ../card1
lrwxrwxrwx. 1 root root system_u:object_r:device_t:s0  13  9.11. 10:44 pci-0000:00:02.0-render -> ../renderD128
```

Besides the SELinux labeling issue, pod files not being symlinks to matching device file might cause issues for some workloads?

mregmi, 2024-03-15:
This is fixed via a change in container-selinux.

closed 2024-03-15 by mregmi