intel/intel-device-plugins-for-kubernetes #1616: Feature to make variations of init container Dockerfiles for other platforms like OCP

mregmi, 2023-12-08:
Currently there is only one variation of Dockerfile for initcontainers. But they differ a bit between vanilla Kubernetes and OpenShift. So we currently modify the Dockerfile to customize for OpenShift. It would be nice if we can do `make dockerfile` to create a separate Dockerfiles for OCP (when run with ENV variable). Some differences are below.
* we use different base image (RedHat UBI)
* We cannot use the currently used toybox as it modifies base image's /bin
* We need some custom commands to install packages as UBI image is small and does not contain everything we need.

mythi, 2023-12-08:
> it is grep, findutils and pciutils

`find` is not covered but there it is needed?

mregmi, 2023-12-11:
Actually find is not needed for this. it was for driver image so not needed here.

mythi, 2023-12-11:
Fixed by #1615

closed 2023-12-11 by mythi