intel/intel-device-plugins-for-kubernetes #1629: Prepare 0.29.0 release

tkatila, 2023-12-18:
Checklist:
 
- [ ] run validation on main
   - [ ]  SGX
   - [ ]  QAT
   - [ ]  DSA
   - [ ]  IAA
   - [ ]  DLB
   - [ ]  FPGA
   - [ ]  GPU
- [ ] create release-0.28 branch
- [ ] make set-version TAG=0.28.0 + commit
- [ ] make lock-images + commit
- [ ] update `publish.yml` to create docs for v0.28
- [ ] draft release notes, review
- [ ] update main README for supported versions and docs URL
- [ ] final validation
- [ ] publish release
- [ ] update main branch's operator CRs to point to 0.28
- [ ] update helm chart
- [ ] update operatorhub.io bundle

mythi, 2024-01-26:
> update operatorhub.io bundle

@hj-johannes-lee can you take care of this?

hj-johannes-lee, 2024-02-08:
https://github.com/k8s-operatorhub/community-operators/pull/3865

closed 2024-02-09 by mythi