intel/intel-device-plugins-for-kubernetes #1642: FPGA: Issue with .aocx bitstream parser

eliaskoromilas, 2024-01-05:
The following statement is not true,

https://github.com/intel/intel-device-plugins-for-kubernetes/blob/v0.29.0/pkg/fpga/bitstream/aocx.go#L31-L32

and parsing FPGA binaries fails (unnecessarily) in many cases:

https://github.com/intel/intel-device-plugins-for-kubernetes/blob/v0.29.0/pkg/fpga/bitstream/aocx.go#L175-L178

kad, 2024-01-05:
In modern AOCX files you've got, what does this field contain? Can you share (e.g. privately) some of example aocx files where it fails parsing?

eliaskoromilas, 2024-01-05:
`intel_a10gx_pac.tar.gz` OpenCL BSP (19.2-57)
* `bringup/aocxs/pac_a10.aocx`: **63b3779b8bdd4f039ceb0301181d6aef**

`intel_s10sx_pac.tar.gz` OpenCL BSP (19.2-57)
* `bringup/aocxs/pac_s10.aocx`: **63b3779b8bdd4f039ceb0301181d6aef**
* `bringup/aocxs/pac_s10_usm.aocx`: 18b79ffa2ee54aa096ef4230dafacb5f

eliaskoromilas, 2024-01-05:
Downloads > oneAPI : https://www.intel.com/content/www/us/en/software-kit/661712/intel-quartus-prime-pro-edition-design-software-version-19-2-for-linux.html