intel/intel-device-plugins-for-kubernetes #1645: VF kernel driver name is c62xvt, not c6xxvt  for C62X device

rzang-intel, 2024-01-08:
Code : https://github.com/intel/intel-device-plugins-for-kubernetes/blob/ff41eb8b42388fb6ff9ad32392d4a9490db5b659/cmd/qat_plugin/dpdkdrv/dpdkdrv.go#L69

According to the latest kernel code:
https://github.com/torvalds/linux/tree/master/drivers/crypto/intel/qat

The VF driver name is c62xvt, not c6xxvt. 
This mistake prevents the plugin from being installed on C62X chipset.

mythi, 2024-01-08:
> The VF driver name is c62xvt, not c6xxvt.
> This mistake prevents the plugin from being installed on C62X chipset.

Even if the driver name says `c62xvf`, it shows up as `c6xxvf` when loaded on the system: `/sys/bus/pci/drivers/c6xxvf/` and that's what this plugin parameter matches. 

What is the error you are seeing?

rzang-intel, 2024-01-08:
Yes, you are right. The driver name did show up as "c6xxvf" in the system. Thanks for correcting me.
The problem I am facing is the intel-qat-plugin container in CrashLoopBackOff state due to (from its log):
[image]

Previously I thought it was unbinding using the wrong driver name. However, it is not the case as you pointed out.
The device 0000:3d:01:0 is the PF device. 
[image]
So could you please shad some light on why it is unbinding the PF device and why it failed?

BTW, I am using release v0.22.0.

rzang-intel, 2024-01-08:
Yes, you are right. The driver did show up as "c6xxxvf" in the system.
The problem I am facing is the intel-qat-plugin container in CrashLoopBackOff state due to (from its log):
[image]
0000:3d:01:0 is the first VF:
[image]

Previously I thought it was because the wrong driver name. However, it is not the case as you pointed out. Could you please also shed some light on why it kept failing on unbinding the VF driver?
BTW, I am using release v0.22.0

rzang-intel, 2024-01-08:
Removed the VF driver. It stops at this error:
[image]

mythi, 2024-01-08:
Are you running Ubuntu? On Apparmor enabled systems the plugin must be deployed using the `unconfined` annotation to avoid Apparmor blocking the sysfs writes. We don't have it properly documented but see https://github.com/intel/intel-device-plugins-for-kubernetes/tree/main/deployments/qat_plugin/overlays/apparmor_unconfined

rzang-intel, 2024-01-08:
Well, it always is beneficial to ask the master! Thanks Mikko! It resolves my problem!

closed 2024-01-08 by rzang-intel