intel/intel-device-plugins-for-kubernetes #1687: QAT

mythi, 2024-03-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I always have to edit this manually to [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

hj-johannes-lee, 2024-03-13:
Maybe, then, we do not need the open pr.
Let me just change the pr to make this issue solved. Do you think it's ok?

mythi, 2024-03-13:
that works OK to me

mythi, 2024-07-26:
>  enable all tests (dpdk compress/crypto + openssl + cpa_sample_code)

@hj-johannes-lee I think we can now finish this final task and add test cases for [QAT Engine](https://github.com/intel/QAT_Engine?tab=readme-ov-file#test-using-openssl-speed-utility). Fox example: a simple k8s `Job` with two containers each running different algo would be sufficient.

mythi, 2024-10-14:
> why we have used the image name `openssl-qat-engine` though the `cpa-sample_code` is a part of qatlib..?

We have maintained a reference container image build for QAT Engine since day 0 but it wasn't possible to add it to the tests until  `e2e-spr`. `cpa_sample_code` worked for basic tests even before `e2e-spr` so we just re-used the reference `openssl-qat-engine` build for it instead for creating a completely new image.

PR #1885 (2024-10-30): QAT, e2e: add qat-engine's testapp case
Fixes: #1687