intel/intel-device-plugins-for-kubernetes #1693: intel-idxd-config-initcontainer cannot use UBI base images.

mregmi, 2024-03-20:
**Describe the support request**
intel-idxd-config-initcontainer cannot be used in OpenShift. The idxd-config initcontainer cannot use UBI minimal images as it is using apt-get to install accel-config. All other Dockerfiles are compatible with UBI. Can we make so that this can also be used with UBI?

**System (please complete the following information if applicable):**
 - OS version: RHEL 9.2 (openshift)
 - Kernel version: Linux 5.15
  - Device plugins version: v0.29.0
 - Hardware info: DSA

mythi, 2024-03-21:
> Can we make so that this can also be used with UBI?

we would not be able to validate (test build) that image since we cannot install accel-config from UBI without the subscription. I guess we could maintain the conditional build flow but we'd need your help to make it suitable for your needs

mregmi, 2024-03-21:
Sure, we are happy to help. we might need help to understand how the conditional flow work in the codebase.

mregmi, 2024-03-29:
We can test this approach and let you know. Thanks

mythi, 2024-04-02:
> Without build arguments, it would generate the debian based version.

This has two problems: 1) the OCP build environment (at least used to) does not allow users to add custom build args so the default needs to be what OCP expects and our builder adds custom args, 2) our builder setup currently knows only about "distroless" base image for all the Go images so some additional logic would be needed to decide when to use distroless vs debian.

closed 2026-01-19 by mythi