intel/intel-device-plugins-for-kubernetes #1706: Intel 14th-gen Meteor Lake

Kiryuumaru, 2024-04-08:
**Describe the support request**
I would like to add new `Intel 14th-gen Meteor Lake` as resources into my pod by:
```
resources: 
  limits:
    gpu.intel.com/i915: 1
```
but pod wont initialize because of lacking resources.

**System (please complete the following information if applicable):**
 - OS version: [e.g. Ubuntu 22.04]
 - Kernel version: [e.g. Linux 5.15]
  - Device plugins version: [e.g. v0.29.0]
 - Hardware info: [e.g. SPR with QAT]

tkatila, 2024-04-08:
Hi @Kiryuumaru could please describe your environment? Which k8s version, which gpu plugin you installed etc.?

tkatila, 2024-04-23:
Closing. Please reopen with more details if this is still an issue.

closed 2024-04-23 by tkatila