intel/intel-device-plugins-for-kubernetes #1710: Not compatible device: version

wmiller112, 2024-04-12:
I have the gpu operator and intel-device-plugin running in a single node Talos cluster running an i5 13600. The operator/plugin were functional until a node (Talos Version) update, and in the process I was adding more drivers for additional peripherals. After reboot of node, the pod from the intel-device-plugin initially exhibited an issue starting up, with the following error
`MountVolume.SetUp failed for volume "nfd-features" : hostPath type check failed: /etc/kubernetes/node-feature-discovery/features.d/ is not a directory`.
Since the DS is generated, I manually updated the volume to `DirectoryOrCreate`, which at least let the pod come up. Once up the verbose logs show:

```
I0412 15:28:48.082670       1 gpu_plugin.go:546] GPU device plugin started with none preferred allocation policy
W0412 15:28:48.082741       1 gpu_plugin.go:303] failed to read by-path dir: open /dev/dri/by-path: no such file or directory
I0412 15:28:48.082761       1 gpu_plugin.go:348] GPU 'i915' resource share count = 3
I0412 15:28:48.082794       1 gpu_plugin.go:388] Not compatible device: version
I0412 15:28:53.082860       1 gpu_plugin.go:388] Not compatible device: version
I0412 15:28:58.086680       1 gpu_plugin.go:388] Not compatible device: version
I0412 15:29:03.085438       1 gpu_plugin.go:388] Not compatible device: version
```

The the plugin does not label the nodes accordingly. I've since reverted all changes to the node back on the original OS version and drivers that were previously working, fully removed and re-installed both operator, plugin, and CRDs, and the error in logs is still the same.

**To Reproduce**
I'm not 100% sure if it was a talos os upgrade or driver change, or something else that may have gotten messed up in the process.

**Expected behavior**
The pod to come up, identify the i915 gpu on the node and label it so pods requesting the gpu can start up

**System (please complete the following information):**
 - OS version: Talos 1.6.7
 - Device plugins version: 0.29.0

**Additional context**
Fairly sure this is something under that got changed under the operator/plugin, but cannot seem to track down what based on the error.

wmiller112, 2024-04-12:
This was a hardware issue. Turned out multi-display  gpu was somehow disabled in bios.

closed 2024-04-12 by wmiller112

tkatila, 2024-04-15:
Hi @wmiller112, thanks for reporting the resolution!

Some background on how the GPU plugin works:
1) GPU plugin labeling nodes (via NFD and /etc/kubernetes/node-feature-rules) is only relevant when used with `resourceManager=true`. Otherwise, node labels are done directly via NFD and its NodeFeatureRules CRD.
2) Plugin detects the device purely via the sysfs entries. So as long as the driver is working, plugin should expose the device to k8s.