intel/intel-device-plugins-for-kubernetes #1826: GPU: i915 resources are not being added to nodes on Talos

djryanj, 2024-09-02:
**Describe the bug**
Talos running on an Intel NUC12WSKv5 with an Intel i5-1250P processor, the resources for `gpu.intel.com/i915` are not added to the nodes. An identical node running k3s shows the resources.

As a result, pods requesting the `gpu.intel.com/i915` cannot be scheduled without removing those requests, and the operator therefore doesn't automatically add the volume mounts for `/dev/dri` to the pod.

Note that adding the volume mounts manually does give correct access to the GPU.

The `gpu.intel.com/device-id` labels exist fine, and are identical, but the resources are not added.

**To Reproduce**
- Install node with Talos, install NFD operator, node will not get resources added to it
- Repeat the same for k3s, node will get resources added to it

**Expected behavior**
Resources should be added to the node correctly.

**Screenshots**
From Talos:

```bash
kubectl describe node talos-nuc-2                                           
Name:               talos-nuc-2
Roles:              worker
Labels:             beta.kubernetes.io/arch=amd64
                    beta.kubernetes.io/os=linux
                    gpu.intel.com/device-id.0300-46a6.count=1
                    gpu.intel.com/device-id.0300-46a6.present=true
                    intel.feature.node.kubernetes.io/gpu=true
                    kubernetes.io/arch=amd64
                    kubernetes.io/hostname=talos-nuc-2
                    kubernetes.io/os=linux
                    kubernetes.io/role=worker
                    node-type=worker
Annotations:        csi.volume.kubernetes.io/nodeid: {"driver.longhorn.io":"talos-nuc-2"}
                    nfd.node.kubernetes.io/feature-labels:
                      gpu.intel.com/device-id.0300-46a6.count,gpu.intel.com/device-id.0300-46a6.present,intel.feature.node.kubernetes.io/gpu
                    node.alpha.kubernetes.io/ttl: 0
                    talos.dev/owned-labels: ["intel.feature.node.kubernetes.io/gpu"]
                    volumes.kubernetes.io/controller-managed-attach-detach: true
CreationTimestamp:  Fri, 28 Jun 2024 11:47:28 -0600
Taints:             <none>
Unschedulable:      false
Lease:
  HolderIdentity:  talos-nuc-2
  AcquireTime:     <unset>
  RenewTime:       Mon, 02 Sep 2024 08:28:49 -0600
Conditions:
  Type                 Status  LastHeartbeatTime                 LastTransitionTime                Reason                       Message    
  ----                 ------  -----------------                 ------------------                ------                       -------    
  NetworkUnavailable   False   Fri, 28 Jun 2024 11:47:47 -0600   Fri, 28 Jun 2024 11:47:47 -0600   CiliumIsUp                   Cilium is running on this node
  MemoryPressure       False   Mon, 02 Sep 2024 08:25:39 -0600   Fri, 26 Jul 2024 16:42:03 -0600   KubeletHasSufficientMemory   kubelet has sufficient memory available
  DiskPressure         False   Mon, 02 Sep 2024 08:25:39 -0600   Fri, 26 Jul 2024 16:42:03 -0600   KubeletHasNoDiskPressure     kubelet has no disk pressure
  PIDPressure          False   Mon, 02 Sep 2024 08:25:39 -0600   Fri, 26 Jul 2024 16:42:03 -0600   KubeletHasSufficientPID      kubelet has sufficient PID available
  Ready                True    Mon, 02 Sep 2024 08:25:39 -0600   Fri, 26 Jul 2024 16:42:03 -0600   KubeletReady                 kubelet is posting ready status
Addresses:
  InternalIP:  <snip>
  Hostname:    talos-nuc-2
Capacity:
  cpu:                16
  ephemeral-storage:  486916820Ki
  hugepages-2Mi:      0
  memory:             15914900Ki
  pods:               110
Allocatable:
  cpu:                15950m
  ephemeral-storage:  448474105114
  hugepages-2Mi:      0
  memory:             15615892Ki
  pods:               110
System Info:
[... 118 lines cut ...]
<snip>
Allocated resources:
  (Total limits may be over 100 percent, i.e., overcommitted.)
  Resource                       Requests      Limits
  --------                       --------      ------
  cpu                            8510m (53%)   5400m (33%)
  memory                         2684Mi (17%)  9928Mi (63%)
  ephemeral-storage              0 (0%)        0 (0%)
  hugepages-1Gi                  0 (0%)        0 (0%)
  hugepages-2Mi                  0 (0%)        0 (0%)
  gpu.intel.com/i915             1             1
  gpu.intel.com/i915_monitoring  0             0
Events:                          <none>
```

**System (please complete the following information):**

 - OS version: Talos v1.7.4 vs Ubuntu 22.04.4 LTS
 - Kernel version: 6.6.32-talos vs 5.15.0-107-generic
 - Kubernetes version: 1.30.0 vs. 1.28.5+k3s1
 - Device plugins version: 0.30.0 vs 0.28.0
 - Hardware info: GPU

**Additional context**
I realize directly comparing the two may be irrelevant, but regardless the resources information does not show up correctly in Talos.

djryanj, 2024-09-02:
Turns out this is not an issue. I neglected to add `pod-security.kubernetes.io/enforce: privileged` to the `inteldeviceplugins-system` namespace as required for device plugin pods to work correctly.

Closing.

closed 2024-09-02 by djryanj

mythi, 2024-09-02:
> I neglected to add pod-security.kubernetes.io/enforce: privileged to the inteldeviceplugins-system namespace as required for device plugin pods to work correctly

We have an issue open to have this properly documented so that users don't run into these type of issues. Perhaps the default deployment could add this automatically...