intel/intel-device-plugins-for-kubernetes #1841: Incomplete plugin SecurityContexts

eero-t, 2024-09-18:
**Describe the bug**

Plugins do not match k8s _baseline_ [security standards](https://kubernetes.io/docs/concepts/security/pod-security-standards/) (mainly due to host mounts), but that could be compensated somewhat by adopting relevant policies from the k8s _restricted_ security standard.   When containers are run with root user, they could at least drop unnecessary privileges...

**To Reproduce**

Check what capabilities are dropped in SecurityContext sections listed in plugins deployments:
https://github.com/intel/intel-device-plugins-for-kubernetes/tree/main/deployments

For example:
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/gpu_plugin/base/intel-gpu-plugin.yaml

**Expected behavior**

Something like this:
```
securityContext:
  readOnlyRootFilesystem: true
  allowPrivilegeEscalation: false
  capabilities:
     drop: [ "ALL" ]
     add: [<whatever is actually necessary>]
  seccompProfile:
    type: RuntimeDefault
```

**Screenshots**

E.g. in above listed GPU plugin deployment, SecurityContext is just:
```
        securityContext:
          seLinuxOptions:
            type: "container_device_plugin_t"
          readOnlyRootFilesystem: true
          allowPrivilegeEscalation: false
```

**System (please complete the following information):**
 - Device plugins version: git HEAD

mythi, 2024-09-24:
@eero-t can you submit a PR for this?

eero-t, 2024-09-24:
> @eero-t can you submit a PR for this?

Not this or next week, maybe after that.

PR #1945 (2025-01-10): depl: drop capabilities from all plugins
Fixes: #1841
closed 2025-01-10 by mythi