intel/intel-device-plugins-for-kubernetes #1851: Prepare 0.31.0 release

tkatila, 2024-09-25:
Checklist:
- [x] run validation on main
   - [x] GNR (IAA, DSA)
   - [x] GNR-D (QAT)
   - [x] FPGA
   - [x] SPR (SGX, QAT, GPU, IAA, DSA)
- [] Make sure `kube-rbac-proxy` is the latest version
- [] create release-0.30 branch
- [] release branch changes
  - [] edit `default_labels.docker` + make dockerfiles
  - [] make set-version TAG=0.30.0 + commit
  - [] update `publish.yml` to create docs for v0.30
- [] draft release notes, review
- [] publish release
- [] main branch changes
  - [] update base README for supported versions and docs URL
  - [] update main branch's operator CRs to point to 0.30, also reconciler.go
- [] update helm chart: [PR](https://github.com/intel/helm-charts/pull/55)
  - [] Make sure to update CRDs and README
- [] update operatorhub.io bundle

mythi, 2024-09-30:
> * update operatorhub.io bundle

we're now back to https://github.com/intel/intel-device-plugins-for-kubernetes/issues/1731#issuecomment-2228091659

tkatila, 2024-10-01:
I added a section about the operatorhub bundle to the release notes.

tkatila, 2024-10-01:
> I added a section about the operatorhub bundle to the release notes.

Nevermind. I'll remove it.

hj-johannes-lee, 2024-10-01:
https://github.com/k8s-operatorhub/community-operators/pull/5088
https://github.com/k8s-operatorhub/community-operators/pull/5091
Done!

mythi, 2024-10-01:
> [k8s-operatorhub/community-operators#5088](https://github.com/k8s-operatorhub/community-operators/pull/5088) [k8s-operatorhub/community-operators#5091](https://github.com/k8s-operatorhub/community-operators/pull/5091) Done!

since this moved to `stable` channel, did you check the upgrade/promotion flows are done correctly?

hj-johannes-lee, 2024-10-04:
https://github.com/intel/intel-device-plugins-for-kubernetes/pull/1865
I found out that actually the upgrade has not been working properly since sha256 is introduced.

closed 2024-10-18 by mythi