intel/intel-device-plugins-for-kubernetes #1963: Prepare 0.32.0 release

tkatila, 2025-01-21:
Checklist:
- [ ] run validation on main
   - [ ] GNR (IAA, DSA)
   - [ ] GNR-D (QAT)
   - [ ] FPGA
   - [ ] SPR (SGX, QAT, GPU, IAA, DSA)
- [ ] create release-0.33 branch
- [ ] release branch changes
  - [ ] edit `default_labels.docker` + make dockerfiles
  - [ ] make set-version TAG=0.33.0 + commit
  - [ ] update `publish.yml` to create docs for v0.33
- [ ] draft release notes, review
  - [ ] Note deprecation for devices
- [ ] publish release
- [ ] main branch changes
  - [ ] update base README for supported versions and docs URL
  - [ ] update main branch's operator CRs to point to 0.33, also reconciler.go
- [ ] update helm chart: 
  - [ ] Make sure to update CRDs and README
- [ ] update operatorhub.io bundle

tkatila, 2025-02-03:
@hj-johannes-lee can you handle the operator bundle?

hj-johannes-lee, 2025-02-11:
@tkatila I am just back from my vacation.!
Let me do right today.!

hj-johannes-lee, 2025-02-11:
done.!

https://github.com/k8s-operatorhub/community-operators/pull/5669

closed 2025-02-12 by tkatila