intel/intel-device-plugins-for-kubernetes #2039: Intel-Gpu-Levelzero: Failed to get Indices from levelZero on WSL2 on Intel Lunar Lake Core Ultra 9 processor

shailesh837, 2025-04-11:
**Describe the bug**
We have Customer called CelebalTech based in Dubai, Finland, Bangalore, Targeting there Customer contact Call center Framework running on AIPC running single node rke2 kuberenetes cluster using OPEA-ERAG based RAG solution.
WSL2 : Ubuntu – 24.04
Document I used to reproduce the issue.
 
Currently we have the Latest the whole OPEA-Based RAG running usign docker-compose on LLM Model : llama3.2 running on iGPU (/dev/dri) exposure for iGPU (Windows driver : version):
After applying side-car (WSL2 : Ubuntu 24-04 gpu device plugin) there is level zero crash or failure inside pod:
 
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. RKE2 version 1.31.4 Installtion
 sudo curl -sfL https://get.rke2.io | sudo INSTALL_RKE2_VERSION=v1.31.4+rke2r1 sh -
2. Enable the rke2-server service
 systemctl enable rke2-server.service
3. Start the service
 systemctl start rke2-server.service
4. Follow the logs, if you like
  journalctl -u rke2-server –f
5) Replace canal cni with calico 
Step 1: Edit this file /etc/rancher/rke2/config.yaml and add cni: calico
Step 2: restart daemonsets and rke2 service
sudo systemctl daemon-reload
sudo systemctl restart rke2-service
WSL layer enables Intel GPU detection with WSL (Windows Subsystem for Linux) Kubernetes clusters.
kubectl apply -k 'https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/gpu_plugin/overlays/wsl?ref=v0.32.0'

**Expected behavior**
A clear and concise description of what you expected to happen.
Intel-gpu-plugin should detect levelzero indices and have gpu pod running under rke2 cluster

[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

**System (please complete the following information):**
 - OS version:  WSL2 : Ubuntu 24.04
 - Kernel version:  microsoft Linux 5.15
 - Device plugins version:  v0.32.0
 - Hardware info: Core Ultra 7 288V

eero-t, 2025-04-11:
Potentially related: https://github.com/intel/compute-runtime/issues/721

tkatila, 2025-04-11:
@eero-t it's not. LunarLake seems to require the igc packages that we have dropped by force in the containers. I have debugged the issue with @shailesh837 today.

shailesh837, 2025-04-17:
@tkatila : : Please let me know when we can have new intel-gpu-levelzero container with this changes in intel docker hub, As its urgently required, we have tested it in UAT with customer the local version we created , But now we need to test on multiple customer machines.

tkatila, 2025-04-22:
@shailesh837 it's in progress. We ran into some issues with our containers which we need to fix first.

closed 2025-04-29 by mythi

tkatila, 2025-04-30:
@shailesh837 the fix was merged yesterday. There is a container available here:
docker.io/intel/intel-gpu-levelzero@sha256:2326d1fb56c93388239786ef8bb7152c3d37dd9ed04a165131211d98d7128831. 

Could you verify that it still works with the LunarLake device? If it does, we'll include the fix in our 0.32.1 release.

tkatila, 2025-05-02:
Noting it also here that @shailesh837 tested the container with the LunarLake device and the detection succeeded.

PR #2043 (2025-04-29): gpu levelzero: update the components to support LunarLake in WSL
Also install the igc-core components as they seem to be required. Sadly the container size will increase with little benefit.

Fixes: #2039