intel/intel-device-plugins-for-kubernetes #2056: Cannot generate operator bundle

chaitanya1731, 2025-05-13:
**Describe the bug**
Unable to generate the operator bundle for tag 0.32.1 using `make bundle' command

**To Reproduce**
Steps to reproduce the behavior:
git clone to v0.32.1 branch and run - `make bundle TAG=0.32.1 CHANNELS=alpha DEFAULT_CHANNEL=alpha`

**Expected behavior**
Operator bundle files generated

**Logs**
```
[ckulkar1@jfz1r09h07 intel-device-plugins-for-kubernetes]$ make bundle TAG=0.32.1 CHANNELS=alpha DEFAULT_CHANNEL=alpha
rm -rf community-operators/operators/intel-device-plugins-operator/0.32.1
mkdir -p community-operators/operators/intel-device-plugins-operator/0.32.1
operator-sdk generate kustomize manifests -q --input-dir deployments/operator/manifests --output-dir deployments/operator/manifests --apis-dir pkg/apis
kustomize build deployments/operator/manifests | sed "s|intel-deviceplugin-operator:devel|intel-deviceplugin-operator:0.32.1|" | operator-sdk generate bundle -q --overwrite --version 0.32.1 --channels=alpha --default-channel=alpha --kustomize-dir deployments/operator/manifests --output-dir . --use-image-digests
Error: accumulating resources: accumulation err='accumulating resources from '../scorecard': evalsymlink failure on '/home/ckulkar1/intel/intel-device-plugins-for-kubernetes/deployments/operator/scorecard' : lstat /home/ckulkar1/intel/intel-device-plugins-for-kubernetes/deployments/operator/scorecard: no such file or directory': evalsymlink failure on '/home/ckulkar1/intel/intel-device-plugins-for-kubernetes/deployments/operator/scorecard' : lstat /home/ckulkar1/intel/intel-device-plugins-for-kubernetes/deployments/operator/scorecard: no such file or directory
2025/05/12 10:29:23 manifests true
2025/05/12 10:29:23 skipping non-yaml file without errors: manifests 
2025/05/12 10:29:23 intel-device-plugins-operator.clusterserviceversion.yaml false
2025/05/12 10:29:23 visited file or dir: "manifests/intel-device-plugins-operator.clusterserviceversion.yaml"
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator:0.32.1 looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator:0.32.1 looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator:0.32.1 looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator:0.32.1 looks like a pullspec
2025/05/12 10:29:23 Found pullspec for annotation intel-deviceplugin-operator-0.32.1-annotation: docker.io/intel/intel-deviceplugin-operator:0.32.1
2025/05/12 10:29:23 Found pullspec for annotation intel-deviceplugin-operator-0.32.1-annotation: docker.io/intel/intel-deviceplugin-operator:0.32.1
2025/05/12 10:29:23 Found pullspec for annotation intel-deviceplugin-operator-0.32.1-annotation: docker.io/intel/intel-deviceplugin-operator:0.32.1
2025/05/12 10:29:23 Found pullspec for annotation intel-deviceplugin-operator-0.32.1-annotation: docker.io/intel/intel-deviceplugin-operator:0.32.1
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator:0.32.1 looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator:0.32.1 looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator:0.32.1 looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator:0.32.1 looks like a pullspec
2025/05/12 10:29:23 manifests/intel-device-plugins-operator.clusterserviceversion.yaml - Replaced pullspec for annotation intel-deviceplugin-operator-0.32.1-annotation: {docker.io intel intel-deviceplugin-operator 0.32.1} -> {docker.io intel intel-deviceplugin-operator sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a}
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator@sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator@sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator@sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator@sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: operators.operatorframework.io/builder:operator-sdk-v1.22.2 looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator@sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator@sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator@sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a looks like a pullspec
2025/05/12 10:29:23 Pull spec heuristic: docker.io/intel/intel-deviceplugin-operator@sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a looks like a pullspec
2025/05/12 10:29:23 manifests/intel-device-plugins-operator.clusterserviceversion.yaml - Set relateImage intel-deviceplugin-operator-ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a-annotation (from annotation intel-deviceplugin-operator-ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a-annotation): docker.io/intel/intel-deviceplugin-operator@sha256:ef1187484172179e6dfeb9dbbc4156958b7ae0661b0ffcbbba3b6b5d04f3e08a
INFO[0001] Creating bundle.Dockerfile                   
INFO[0001] Creating metadata/annotations.yaml           
INFO[0001] Bundle metadata generated suceessfully       
# Remove unneeded resources
rm manifests/*service.yaml
rm: cannot remove 'manifests/*service.yaml': No such file or directory
make: *** [Makefile:126: bundle] Error 1
```

**Additional context**
Got same issue on v0.32.0

chaitanya1731, 2025-05-13:
I even tried updating operator-sdk but then got this new error - 
```
[ckulkar1@jfz1r09h07 intel-device-plugins-for-kubernetes]$ make bundle CHANNELS=alpha DEFAULT_CHANNEL=alpha
rm -rf community-operators/operators/intel-device-plugins-operator/0.32.1
mkdir -p community-operators/operators/intel-device-plugins-operator/0.32.1
operator-sdk generate kustomize manifests -q --input-dir deployments/operator/manifests --output-dir deployments/operator/manifests --apis-dir pkg/apis
Error: no plugin could be resolved with key "go.kubebuilder.io/v2" for project version "3"
Usage:
  operator-sdk [flags]

Examples:
The first step is to initialize your project:
    operator-sdk init [--plugins=<PLUGIN KEYS> [--project-version=<PROJECT VERSION>]]

<PLUGIN KEYS> is a comma-separated list of plugin keys from the following table
and <PROJECT VERSION> a supported project version for these plugins.

                             Plugin keys | Supported project versions
-----------------------------------------+----------------------------
     ansible.sdk.operatorframework.io/v1 |                          3
 deploy-image.go.kubebuilder.io/v1-alpha |                          3
                    go.kubebuilder.io/v4 |                          3
         grafana.kubebuilder.io/v1-alpha |                          3
        helm.sdk.operatorframework.io/v1 |                          3

For more specific help for the init command of a certain plugins and project version
configuration please run:
    operator-sdk init --help --plugins=<PLUGIN KEYS> [--project-version=<PROJECT VERSION>]

Default plugin keys: "go.kubebuilder.io/v4"
Default project version: "3"

Flags:
  -h, --help                     help for operator-sdk
      --plugins strings          plugin keys to be used for this subcommand execution
      --project-version string   project version (default "3")

FATA[0000] no plugin could be resolved with key "go.kubebuilder.io/v2" for project version "3"
make: *** [Makefile:123: bundle] Error 1
[ckulkar1@jfz1r09h07 intel-device-plugins-for-kubernetes]$
```

chaitanya1731, 2025-05-13:
Just realized that PR https://github.com/intel/intel-device-plugins-for-kubernetes/pull/1986 - was not added in release v0.32.1.

Can you please check @tkatila @hj-johannes-lee @mythi

chaitanya1731, 2025-05-13:
Update - manually added changes from #1986  worked for `v0.32.1` bundle

mythi, 2025-05-14:
> Update - manually added changes from [#1986](https://github.com/intel/intel-device-plugins-for-kubernetes/pull/1986) worked for `v0.32.1` bundle

@chaitanya1731 is this enough for you to proceed?

tkatila, 2025-05-14:
I suppose we could cherry-pick the fix to the `release-0.32` branch if necessary. I wouldn't move or create another tag, though.

mythi, 2025-05-20:
Fixed by  #2058

closed 2025-05-20 by mythi