intel/intel-device-plugins-for-kubernetes #2063: Device Plugins Operator does not have an ARM image...

jjacobsson, 2025-05-24:
When deploying the Intel Device Plugin Operator with the new Operator Lifetime Manager (why did you do this? nvm, not sure I want to know) the resulting deployment for idpo does not come with a nodeSelector at all. Meaning the operator might end up being scheduled to whatever hardware in the cluster. But the container image appears to only be built for x86 architectures.

And since I now have to use OLM there appears no way for me to add a nodeSelector as olm just overwrites the deployment if I edit it. 

Also most of your documentation on how to deploy the idpo is now incorrect as it is not updated to point at the olm install, only the repository readme mentions it as far as I have found.

jjacobsson, 2025-05-24:
I managed to fix this by editing the deployment for `intel-deviceplugins-controller-manager` and changing `olm.managed:` to false. Then I could add a nodeSelector without it getting overwritten by OLM...

jjacobsson, 2025-05-24:
After much trial and error and searching for documentation I found out that I can edit the "Subscription" thus:
```
apiVersion: operators.coreos.com/v1alpha1
kind: Subscription
metadata:
  name: intel-device-plugins-operator
  namespace: operators
spec:
  channel: stable
  name: intel-device-plugins-operator
  source: operatorhubio-catalog
  sourceNamespace: olm
  config:
    nodeSelector:
      kubernetes.io/arch: amd64
```

The new bit is the `config:` section. This gets added in the deployment which in turn forces the operator to run on compatible hardware in my cluster.

So I did fix this myself, but your installation documentation leaves _much_ to be desired.

closed 2025-05-24 by jjacobsson

tkatila, 2025-05-25:
Hi @jjacobsson, very sorry for the inconvenience. Thanks for reporting back with the fix so we can add it to our documentation!