intel/intel-device-plugins-for-kubernetes #2064: sgx-aesmd-demo image build failing for 0.32.0 & 0.32.1 release versions

raghav19, 2025-05-26:
**Describe the bug**
sgx-aesmd-demo build fails for 0.32.0 and 0.32.1 release version

**To Reproduce**
With latest code , unable to build sgx-aesmd-demo

```shell
cd /tmp/
git clone --branch v0.32.1 https://github.com/intel/intel-device-plugins-for-kubernetes.git

cd intel-device-plugins-for-kubernetes
make sgx-aesmd-demo
```

**Expected behavior**
sgx-aesmd-demo docker image should be built successfully

**Screenshots**

[image]

**System (please complete the following information):**
 - OS version: Ubuntu 22.04
 - Kernel version: 5.15.167.4-microsoft-standard-WSL2
 - Device plugins version: NA
 - Hardware info: WSL2
 -
**Additional context**
Add any other context about the problem here.

mythi, 2025-05-26:
Our `Dockerfile` does not pull from those URLs but they are defined by the repository metadata. It seems broken at the moment.

vinit-intel, 2025-05-28:
Thanks for the fix @mythi . Able to build the sgx-aesmd-demo now.